motioneye-project/motioneye #2768: Snapshot URL on works on second attempt

Rolleroo, 2023-05-20:
I'm having issues pulling the snapshot

First attempt at snapshot fails, second request for snapshot is successful. Iif I have a browser window logged into Motioneye then snapshots work first time.

Request is being made using the following call:
http://192.168.4.16:8765/picture/1/current/?action=snapshot

Authentication for streaming has been disabled.

Any assistance much appreciated.

[image]

MichaIng, 2023-05-21:
Does the service return any error?
```sh
journalctl -u motioneye
```
Are you using motionEye 0.42.1 or 0.43.0 (`dev`)?

Rolleroo, 2023-05-21:
Running 0.43.0.

Not sure I can read these right but is that an authentication error?

[Journal.txt](https://github.com/motioneye-project/motioneye/files/11525172/Journal.txt)

Rolleroo, 2023-05-26:
OK so I ended up going to this repo for my solution. Never worked out what was going wrong and I didn't need to fine grain controls of Motioneye.

https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/output_http/README.md

closed 2023-05-26 by Rolleroo