motioneye-project/motioneye #2769: Streaming URL doesn't work

Tamas92, 2023-05-20:
I'm running the motioneye via docker compose. Everything works fine except the streaming URL. If I go to the http://myip:8081 it shows nothing. (This site can't be reached) Everything else is working, snapshot url etc.  I checked the netstat -tulpn | grep 8081, no output unfortunately. What can be the problem?

MichaIng, 2023-05-21:
Where did you pull the container from? Obviously that `motion` port needs to passed between container and host via `docker run` argument or Docker Compose YAML.

Tamas92, 2023-05-21:
I'm using the ccrisan/motioneye:master-amd64. 

Anyway here is my docker compose file:

```
version: "3"
services:
  motioneye:
    image: ccrisan/motioneye:master-amd64
    container_name: motioneye
    restart: always
    user: root
    ports:
      - 8765:8765
      - 8081:8081
    devices:
      - /dev/video2:/dev/video2
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./etc:/etc/motioneye
      - ./lib:/var/lib/motioneye
      - ./var/run/motion:/var/run/motion
```

MichaIng, 2023-05-21:
You might want to try the new version form here: https://github.com/motioneye-project/motioneye/pkgs/container/motioneye

Is this motion process probably using another port? Not sure how to check those within the container, probably the full command contains it, like shown in `htop` (on the host) or similar.

Tamas92, 2023-07-12:
I bought another server and the streaming is working properly on it with the same docker compose config.

closed 2023-07-12 by Tamas92