motioneye-project/motioneye #2771: Significantly higher CPU usage with motioneye Docker image 0.43.0 over 0.42.1 (Raspberry Pi)

jose1711, 2023-05-24:
Raspberry Pi 3B running Rapsbian 11 Bullseye, USB camera + two external camera streams. MotionEye running in a Docker container.
  * legacy image: MotionEye 0.42.1, Motion 4.2.2 - `ccrisan/motioneye:master-armhf`
  * current image: MotionEye 0.43.0, Motion 4.5.1 - `ghcr.io/motioneye-project/motioneye:edge`

The configuration is identical, yet when no client is connected the legacy container consumes approx. 3-4 % CPU (checked using `docker stats`) vs 25-32 % CPU with the current MotionEye version. The most prominent processes are 3 `motion` processes taking approx. 30 %, 13 % and 10 % respectively - in legacy version `motion` processes usually do not exceed 5 %.

Is there anything that could explain this behaviour (perhaps more recent versions of `motion` are more CPU demanding)?

zagrim, 2023-05-25:
Are the camera configs equal (apart from the config parameter changes from Motion 4.2 to 4.5)? Could you compare the camera-*.conf files from both cases, or attach them here (with any secrets redacted)?

jose1711, 2023-05-25:
> Are the camera configs equal (apart from the config parameter changes from Motion 4.2 to 4.5)? 

Did not make any changes - using the same files for both containers.

I've reduced cameras to a single remote camera served from another appliance. Files for test case: [docker_files.zip](https://github.com/motioneye-project/motioneye/files/11563988/docker_files.zip) - (internal) IP is redacted.

This is how a test is executed:
- collect data from the current Docker image:
```
docker rm benchmark-latest
docker build -t benchmark-latest -f Dockerfile-latest .
docker run --name=benchmark-latest -p 8765:8765 -ti benchmark-latest
for i in $(seq 1 30); do docker stats --no-stream --format "{{.CPUPerc}}" benchmark-latest | tr -d '%'; done
for i in $(seq 1 30); do docker stats --no-stream --format "{{.NetIO}}" benchmark-latest; done | awk '{sub(/MB/, ""); print ($1-a); a=$1}'
```
  - collect the data using legacy Docker image
```
docker rm benchmark-042
docker build -t benchmark-042 -f Dockerfile-042 .
docker run --name=benchmark-042 -p 8765:8765 -ti benchmark-042

for i in $(seq 1 30); do docker stats --no-stream --format "{{.CPUPerc}}" benchmark-042 | tr -d '%'; done
for i in $(seq 1 30); do docker stats --no-stream --format "{{.NetIO}}" benchmark-042; done | awk '{sub(/MB/, ""); print ($1-a); a=$1}'
```
During the test no client (HTTPS) connection is made towards MotionEye server

  - results are visualized on a chart. Note that current (latest) version also generates more inbound traffic.

## CPU%
[image]

## Network traffic inbound (MB/s)
[image]

jose1711, 2023-05-25:
It seem as if the newer version of `motion` grabs the images from a remote (network) camera the whole time - even if there is no real need (motion detection/saving is off and no client is requesting viewing).

jose1711, 2023-05-25:
Now I think this has more to do with updates to `motion` itself, therefore I filed https://github.com/Motion-Project/motion/issues/1673 and closing this one.

closed 2023-05-26 by jose1711