motioneye-project/motioneye #2776: Recent software upgrade lead to corrupted preview images

deelkar, 2023-06-09:
I'm using 4 raspberry pis as camera devices. The video is collected on a central hub (x86 based PC) which has had a recent software upgrade. Since then the video streams are still fine, but the browser view shows corrupted images. 
Recordings transcoded to MP4 also work and look fine.
Manual snapshots are configured but don't get saved to disk.
motioneye.log is not helpful.
Where would I need to look to fix that?

srmeister, 2023-06-12:
there is a bug in libmicrohttpd12. So either downgrade that or use the dev version of motioneye which has a workaround

spupuz, 2023-07-03:
i'm on the latest dev but still get preview corrupted for videos

JamBalaya56562, 2026-11-12:
This is the preview corruption of #2401 and #2871: newer libmicrohttpd versions send Motion's stream chunked, which motionEye's MJPEG client did not handle. #2432 makes motionEye ask Motion for the stream with HTTP/1.0, so it is not chunked; it is in 0.43.1, and MichaIng confirmed the fix on #2871. Closing as fixed; if previews are still corrupted on 0.43.1 or later, please open a new issue.

closed 2026-11-12 by JamBalaya56562