motioneye-project/motioneye #2779: Motioneye & Raspberry PI4B 8 Gb

ita3bc, 2023-06-13:
Fresh install on Raspbian 11 64 bit Desktop. Adding even a single ipcam the average CPU used is almost 90% (2 cores are always at 100% activity), with the consequence that the temperature rises very much in a very short time. This is not good for a system that should work 24/7, is it normal? Is there any way to fix it? I like this software and would hate to abandon it for this reason. Can anyone give me a solution? Thank you
Bye
Fred

zagrim, 2023-06-21:
It depends a lot on the camera resolution, frame rate and your MotionEye/Motion config. But basically I don't think even RPi4 can have good performance if decoding/encoding of video is done in CPU and not in GPU.
You might want to check other related issues on the topic: https://github.com/motioneye-project/motioneye/issues?q=is%3Aissue+cpu+100%25

I think #1051 for example provides some general info and advice, even though the OP was not using RPi but something beefier.

As for how to utilise GPU, that might also be somewhat tricky with RPi, which surely also has been discussed here before. I'm sorry for being vague but there's usually no silver bullet. The easiest way is to reduce your requirements for video quality and overlays (if you have them now).

JamBalaya56562, 2026-11-12:
zagrim's answer covers it: with a network camera, Motion decodes every frame on the CPU, so the load mostly follows the camera's resolution and frame rate. Lowering those in the camera itself (or using its sub-stream), and lowering the frame rate in motionEye, is what brings it down; #1051 has more on this. As there was no follow-up since 2023, closing as answered.

closed 2026-11-12 by JamBalaya56562