motioneye-project/motioneye #2780: Please update installation instructions for Debian 13 bookworm

hakova, 2023-06-15:
Dear developers,

Thank you for maintaining this wonderful project. I recently upgraded my server to Debian 13 bookworm, which does not allow installing motioneye module via pip as the instructions suggest. I was able to create a virtual environment and make changes to the service file to get it work, but I think updated instructions would help people like me a lot. Thank you again so much!

spindles7, 2023-07-11:
Hi hakova,
I have a similar problem - I had motioneye installed and working in Debian 11 (Buster) but after the upgrade to Bookworm the Motionye web interface is not working properly - one camera is OK, but some don't show at all and others have flashing images.    I assume some incompatibility with the new versions of python?    Would be graet to know how you got it working.   Many thanks,

spindles7, 2023-07-11:
Soory, meant Bullseye, not Buster

CryingByte, 2023-07-21:
@hakova @haruzept 
for installation instruction look at #2778 
> Check out the `dev` branch: https://github.com/motioneye-project/motioneye/tree/dev#installation

if you get a error like _error: externally-managed-environment_ on step 2 remove file `/usr/lib/python3.x/EXTERNALLY-MANAGED`
(replace 3.x with your python version e.g 3.11, also below)

after installation check status with
```
systemctl status motioneye
```
if you have installed python > 3.9 and get an error like _AttributeError: module 'collections' has no attribute 'MutableMapping'_ edit file `/usr/local/lib/python3.x/dist-packages/tornado/httputil.py` and change
```python
class HTTPHeaders(collections.MutableMapping):
```
to
```python
class HTTPHeaders(collections.abc.MutableMapping):
```
see https://docs.python.org/3.9/library/collections.html#module-collections

I addtionally got an error *CRITICAL: log directory "/var/log"* does not exist or is not writable and changed the `log_path` in `/etc/motioneye/motioneye.conf` to `/var/log/motioneye`

Hope I could help

Pedder007, 2023-10-31:
Hi @CryingByte, I just updated my Motioneye Proxmox Container to Debian 12 (Bookworm). After doing so Motioneye in general works, but in the WEB GUI I have some colored flickering in the streams of two of my three cams. It's only in the bottom of the streams, roughly 1/6 of the entire video.
Searching for a solution I followed the steps above regarding the occuring error ' externally-managed-environment' and in general (with some warnings about running python as root) the installation worked after having deleted the EXTERNALLY-MANAGED file in the python3.11 folder.
But afterwards my status looked like this and Motioneye doesn't start at all:
````
motadm@motioneye:/$ systemctl status motioneye
x motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Tue 2023-10-31 22:34:56 CET; 5min ago
   Duration: 39ms
    Process: 329 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
   Main PID: 329 (code=exited, status=1/FAILURE)
        CPU: 34ms
`````
Following further on but found, that in '/usr/local/lib/python3.11/dist-packages' there is no '.../tornado/httputil.py'.

Do you have any further ideas/help on this?
(for the first I will restore my Bullsyey backup into the container and continue using this)

Pedder007, 2023-11-01:
@CryingByte and others: I went now for another approach in my Proxmox container:
I installed the container completely new from scratch (on basis of the Proxmox Debian 12 standard template) and then followed the complete installation guide:
https://github.com/motioneye-project/motioneye/tree/dev#installation

After that I went along possible issues explained here and fixed them:
https://github.com/motioneye-project/motioneye/issues/2780

In my case (Proxmox) I had an additional issue when trying to start motioneye, which told me
when typing ```` systemctl status motioneye ````
, that motioneye is not able to write into the
```` /var/run```` directory, which is needed to store the 'motion.pid' file.

Thus I added a directory called ````/motioneye```` into the ````/run```` directory and gave it all rights by
```` chmod 777 motioneye````

Additionally I adjusted the respective path in the ````/etc/motioneye/motioneye.conf```` into ````run_path /var/run/motioneye````

When now starting the first time, the motion-user directly occupied the directory for himself and smoothly stored it's motion.pid file into it.

I quickly rebooted the container (just to be sure about not having kept 'somewhere' 'something' 😄 ) and now motioneye runs smoothly and shows my video streams :-) on basis of debian bookworm

eckerj, 2024-01-03:
> Hi @CryingByte, I just updated my Motioneye Proxmox Container to Debian 12 (Bookworm). After doing so Motioneye in general works, but in the WEB GUI I have some colored flickering in the streams of two of my three cams. It's only in the bottom of the streams, roughly 1/6 of the entire video. 

I've been having the same issue and found it related to libmicrohttpd12.  On my ubuntu 22.04, the "unattended-upgrade" kept re-installing the "bad" version (0.9.75-3ubuntu1), but was finally able to "blacklist" it from being installed again by using:

apt-mark hold libmicrohttpd12

My flickering/changing stream colors (in my case, it was 2/3rds of the stream) went away after downgrade-installing the 0.9.72-2 version of libmicrohttpd12 and restarting motioneye.

Hope this helps.

MichaIng, 2024-01-23:
Fixed with: https://github.com/motioneye-project/motioneye/commit/2862efe
For other issues, like with `libmicrohttpd12`, please open a new issue, in case there does not exist one already (which IIRC does).

closed 2024-01-23 by MichaIng