motioneye-project/motioneye #2785: recieving blank attachements on telgram notifications

orbit80, 2023-07-02:
Hello ,
the issue is pretty simple 
telegram notifactions are working , i do get the test message from the TG boot 
but only a few motion images are beeing sent successfully

most of the messages are just : 
Motion has been detected by camera "Office/DietPi" at 2023-06-29 20:49:54 (local time).

only a few motion images are attached successfully.

my configuration : 
attached images time : 30
no email notifications
no web hooks
no running commands

on still images section :
enable manual snapshots - off 
capture mode - auto

zagrim, 2023-07-02:
There is no such capture mode as "auto", at least not in the current dev or rather ancient 0.42-rc1 which are the ones I had readily available for testing. Do you get any saved pictures for the camera when motion is triggered (please check using the web interface)?

Anyway, I just tested Telegram notification with current dev branch (still image capture mode = "Motion triggered", attached images time for Telegram = 30) and I did get plenty of pics (well, plenty of messages with one pic per message).

orbit80, 2023-07-04:
Thanks zagrim
indeed , the problem was with  capture mode definition , it was configured as  "Motion Triggered [ One Picture ] "
after changing it to "Motion triggered" , it starts to send bursts of images every single motion detedcted event.

as far as i concern , it fullfils my demands .
so thanks again , appriciate it.

closed 2023-07-04 by orbit80