motioneye-project/motioneye #2791: BeaglePlay

Tresillo2017, 2023-07-12:
I'm using a dev build of motion eye os. I was wondering if it's posible to use the dedicated video processor of the beagle play (ARM Cortex M4F). Video is laggy since the main processor isn't build for Realtime video processing.

> More info [here](https://git.beagleboard.org/beagleplay/beagleplay)

zagrim, 2023-07-15:
As for encoding video (i.e. saving of motion videos) you might get improvement by manually rebuilding ffmpeg (which is being used for assembling motion videos) by following these instructions (with adjustments for your case): https://github.com/motioneye-project/motioneye/issues/930#issuecomment-522848452
In that issue the topic was to enable h264_omx, what you seem to be after is something else, but as long as ffmpeg has support for whatever hardware acceleration you need, that should work. I have no experience of BeagleBoards and can't know how long the rebuild might take, though, or if what you are asking for is supported by ffmpeg, or if you need some additional stuff to accomplish what you need. But I hope you can use those instructions as a starting point anyhow.

zagrim, 2023-07-15:
Oh, and if you really are using MotionEyeOS (as you wrote), please note that this is a different project (MEOS is a full operating system prepackaged with all that is needed for running MotionEye - [Github project](https://github.com/motioneye-project/motioneyeos)). This github project if for the standalone version of MotionEye.
If you are using MEOS you will likely not be able to rebuild anything on the device, since there is no easy way of pulling in all the required tools since MEOS is not meant to be a multi-purpose OS.

Tresillo2017, 2023-07-15:
> Oh, and if you really are using MotionEyeOS (as you wrote), please note that this is a different project (MEOS is a full operating system prepackaged with all that is needed for running MotionEye - [...]

My bad, I wrote MotionEyeOS instead of MotionEye, i'm running the pip package.

And the GPU's name is [PowerVR® Rogue™ AXE-1-16 GPU](https://www.imaginationtech.com/product/img-axe-1-16/)

zagrim, 2023-07-16:
That page lists Vulcan 1.3 as being supported, and ffmpeg docs also mention Vulkan support ([API implementation status](https://trac.ffmpeg.org/wiki/HWAccelIntro#FFmpegAPIImplementationStatus), [some specifics](https://trac.ffmpeg.org/wiki/HWAccelIntro#Vulkan)), so it should be possible to get some level of GPU support working for decoding of video with a recent version of ffmpeg, at least after building it with Vulkan support if your Linux distribution doesn't come with it already built in.

zagrim, 2023-07-16:
As for how to actually get Motion utilising Vulkan support, I'm not certain. Looking at Motion docs it looks like [`decoder` option under `netcam_params`](https://motion-project.github.io/motion_config.html#netcam_params) might allow that if your camera is a network camera (connects using http(s)/rtsp/rtmp). First, you need to figure out the decoder name by executing `ffmpg -decoders` - see below quote from Motion docs. Then it should be possible to use by entering `netcam_params decoder=<DECODER NAME FOR VULKAN>` to "Extra Motion Options".

> Users can find the names of the decoders in ffmpeg on their system by going to a command prompt and typing ffmpeg -decoders and looking for the name of the decoder with a 'D' next to it and correlating this with the type of image provided from the camera. Note that although the decoder may be listed, it still may not work within Motion.

JamBalaya56562, 2026-11-12:
Thanks zagrim for the details. To sum up: motionEye itself does no video processing; Motion decodes the camera's frames and ffmpeg encodes the movies. So using the BeaglePlay's GPU depends on Motion and ffmpeg supporting it, e.g. a Vulkan decoder through `netcam_params decoder=…` in "Extra Motion Options" for a network camera. As there was no follow-up since 2023, closing as answered.

closed 2026-11-12 by JamBalaya56562