motioneye-project/motioneye #2793: MotionEye Only Starts Once

Ryu945, 2023-07-14:
I installed Raspberry Pi OS 11.  I then went through these step and got MotionEye working.  https://jmichault.github.io/motioneye.eo-dok/en/instalado_en_debian/

I brought up the GUI just fine by going to ip:8765

The issue is that I cannot get it to ever come up again when I reboot.   I found all the folder this guide said I would have to make  myself already made.   https://trevilly.com/installation-de-motioneye-sur-raspbian-buster-videosurveillance/

Any idea how to fix this?

Ryu945, 2023-07-18:
I couldn't get meyectl to start under the motion user with the terminal command but I can it get started if I make it a systemd service.  The problem I run into is that upon launching the program it crashes.  Where would I find logs on why it is crashing?

zagrim, 2023-07-19:
If you wish to continue this here, I'd suggest you to switch to motioneye-project (this github project) and refer to the installation instructions provided [here](https://github.com/motioneye-project/motioneye/tree/dev) instead of that fork you are now using, since there are some differences due to the fork not being completely up-to-date. For example, the installation script has been changed somewhat [here](https://github.com/motioneye-project/motioneye/commits/dev/motioneye/extra/linux_init) as compared to the [fork](https://github.com/jmichault/motioneye.eo/commits/master/motioneye/extra/linux_init). Mostly those differences seem to be about Motion 4.4 support, but there might be some relevance. And I think that the installation should end up having the systemd service already in place (so no need for calling meyectl manually).

Ryu945, 2023-07-23:
Thankyou.  I'll try that and move future tickets to their.   I have tried running it on Debian 11 64 bit for Raspberry Pi 4B and I noticed the software run fine and starts on boot okay but the camera is not working.

Ryu945, 2023-07-31:
Using Debian 10 fixed this problem.

This is the MotionEye install that worked for me.

https://github.com/motioneye-project/motioneye/tree/dev

closed 2023-07-31 by Ryu945