motioneye-project/motioneye #2794: How can I send telegram notifications to an specific topic?

Tresillo2017, 2023-07-16:
Telegram notifications are sent to the default channel inside a topics group

fxb666fdfds, 2026-11-12:
Telegram needs a separate **`message_thread_id`** in addition to the group's `chat_id`. In the [current motionEye sender](https://github.com/motioneye-project/motioneye/blob/e765d633f86d77c064e6bc13c4c028d6a8de47f5/motioneye/sendtelegram.py), the text request contains only `chat_id` and `text`; the photo request also has no topic field. Changing the chat ID alone therefore cannot select a forum topic.

A workaround for **text-only** alerts is Motion Notifications → **Run A Command**, calling a small curl configuration file. For example, save this as `/etc/motioneye/telegram-topic.curl`, replacing the three placeholder values:

```text
url = "https://api.telegram.org/botREPLACE_WITH_BOT_TOKEN/sendMessage"
data-urlencode = "chat_id=REPLACE_WITH_GROUP_ID"
data-urlencode = "message_thread_id=REPLACE_WITH_TOPIC_ID"
data-urlencode = "text=Motion detected"
fail
silent
show-error
max-time = 15
```

Set the notification command to:

```sh
curl --config /etc/motioneye/telegram-topic.curl
```

Keep that file readable only by the account running the notification (it contains the bot token); `curl` and the file must be available inside the container if using Docker. The group ID is normally negative, whereas the topic ID is an integer available as `message_thread_id` on an incoming message from that topic. The bot also needs permission to post there.

This follows Telegram's [sendMessage parameters](https://core.telegram.org/bots/api#sendmessage). It sends an independent text alert, so it does not include motionEye's photo attachments. Native support would need to carry the topic ID through both text and photo requests.

AI disclosure: OpenAI Codex checked the source and API documentation; no live Telegram message was sent to validate this configuration.

JamBalaya56562, 2026-11-12:
Thanks for checking. Confirmed on the current `dev`: `sendtelegram.py` sends no `message_thread_id`, neither with the text nor with the pictures, so a topic cannot be chosen today. I am keeping this open as a feature request: an optional topic ID in the Telegram settings, passed to both requests.