motioneye-project/motioneye #2798: MMAL camera - An error occured. Refreshing is recommended.

SpongeBob-RoundPants, 2023-07-25:
I have a problem adding my Camera module in Motioneye.
When I try to add a local MMAL camera, I get an error message saying 
"An error occured. Refreshing is recommended."

My setup is:
- PI 4, 8GB
- latest Bullseye OS 32bit with desktop
- Camera module v3
- Motioneye installed according to the instructions at https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye

The camera module is connected correctly and working with "libcamera-hello" command, but I can't seem to be able to hook it up together with Motioneye. When I connect a regular USB camera to Motioneye, that one works fine however.

Does anyone have an advice for a beginner like me on how to fix it?

KernelBypass, 2023-07-28:
same issue, but on 64 bit OS

zagrim, 2023-07-29:
Libcamera support is still "under construction", and that applies to the yet unreleased ME 0.43 with Python3 support, @SpongeBob-RoundPants the instructions you refer are for the old Python2 version. Installation instructions for Python3 version are here: https://github.com/motioneye-project/motioneye/tree/dev#installation
Libcamera support has been discussed here: #2425 (and here: #2683)

As far as I've understood things correctly, there are some workarounds mentioned in #2683, but no complete solution yet. There is a PR (#2765) linked to the other issue and it is waiting for testing, with some instructions on how that could be done (but I admit they might not be good instructions in case one is not that tech-savvy).

closed 2026-11-01 by JamBalaya56562