motioneye-project/motioneye #2799: MotionEye mit RPi Camera  Modul3

Schnitzer50, 2023-07-30:
Hallo,
für mein nächstes Projekt will ich das neue RPi Camera Modul3 Noir einsetzen.

Dazu habe ich die aktuelle Version von dietpi mit motioneye auf Basis Python3 installiert und mit einer USB-Cam sowie einem RPi Camera Modul V1.3 erfolgreich getestet.

Leider kann ich unter motioneye das RPi Camera Modul V3 nicht ansprechen.
Welche der angegebenen Kameratypen muss ich  in der Configuration auswählen, es werden die gleichen Auswahlen mehrfach angezeigt?

Für einen Hinweis wäre ich sehr dankbar.

Mit freundlichen Grüßen
W. Stock

Schnitzer50, 2023-08-16:
Hello,
I want to use the new RPi Camera Modul3 Noir for my next project.

For this I installed the current version of dietpi with motioneye based on Python3 and successfully tested it with a USB cam and an RPi Camera Module V1.3.

Unfortunately I cannot address the RPi Camera Module V3 under motioneye.
Which of the specified camera types do I have to select in the configuration, are the same selections displayed multiple times?

I would be very grateful for a hint.

Best regards
W Stock

zagrim, 2023-08-23:
I wonder if that requires the newer libcamera stack... If so, all I can say is that libcamera support is still "under construction", and that applies to the yet unreleased ME 0.43 with Python3 support. The topic has been discussed here: https://github.com/motioneye-project/motioneye/issues/2425 (and here: https://github.com/motioneye-project/motioneye/issues/2683).

As far as I've understood things correctly, there are some workarounds mentioned in https://github.com/motioneye-project/motioneye/issues/2683, but no complete solution yet. There is a PR (https://github.com/motioneye-project/motioneye/pull/2765) linked to the other issue and it is waiting for testing, with some instructions on how that could be done (but I admit they might not be good instructions in case one is not that tech-savvy).

JamBalaya56562, 2026-10-31:
The Camera Module 3 works with current motionEye through `libcamerify`, following the steps in #2812: checked on a Raspberry Pi 5 with a Camera Module 3 and motion 4.7.1. Native libcamera support is tracked in #2812, so closing this as a duplicate of it. If the camera still does not work with these steps, please reopen.

closed 2026-10-31 by JamBalaya56562