motioneye-project/motioneye #2802: Still good in 2023?

Gypzyp, 2023-08-01:
Can we still get motioneye following this?

I get this error:
sudo pip2 install motioneye
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Collecting motioneye
  Using cached motioneye-0.42.1.tar.gz (758 kB)
Collecting tornado<6,>=3.1
  Using cached tornado-5.1.1.tar.gz (516 kB)
Collecting jinja2
  Using cached Jinja2-2.11.3-py2.py3-none-any.whl (125 kB)
Collecting pillow
  Using cached Pillow-6.2.2-cp27-cp27mu-manylinux1_x86_64.whl (2.1 MB)
Collecting pycurl
  Using cached pycurl-7.43.0.5.tar.gz (216 kB)
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-19pZPG/pycurl/setup.py'"'"'; __file__='"'"'/tmp/pip-install-19pZPG/pycurl/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-OAVWU6
         cwd: /tmp/pip-install-19pZPG/pycurl/
    Complete output (11 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-19pZPG/pycurl/setup.py", line 961, in <module>
        ext = get_extension(sys.argv, split_extension_source=split_extension_source)
      File "/tmp/pip-install-19pZPG/pycurl/setup.py", line 623, in get_extension
        ext_config = ExtensionConfiguration(argv)
      File "/tmp/pip-install-19pZPG/pycurl/setup.py", line 101, in __init__
        self.configure()
      File "/tmp/pip-install-19pZPG/pycurl/setup.py", line 238, in configure_unix
        raise ConfigurationError(msg)
    __main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

Can someone please guide me where I can install motioneye in 2023 on Ubuntu. I tried it with docker but no success.

zagrim, 2023-08-01:
There is yet unreleased Python3 version of MotionEye (with plenty of other improvements, too) here: https://github.com/motioneye-project/motioneye/tree/dev#what-is-motioneye

acyanic, 2023-11-08:
The docker readme.md info on this site is a tad old.

    legacy image: MotionEye 0.42.1, Motion 4.2.2 - ccrisan/motioneye:master-armhf
    current image: MotionEye 0.43.0, Motion 4.5.1 - ghcr.io/motioneye-project/motioneye:edge

JamBalaya56562, 2026-11-04:
motionEye runs on Python 3 since 0.43.1 (#1978), so `pip2` is no longer used: the installation steps are in the README. Checked with 0.44.0 on Python 3.10 in a container, in a venv: `pip install motioneye` works without `curl-config`, as pycurl comes as a wheel, and the server starts. Not checked on Ubuntu itself. Closing as completed; please reopen if the README steps fail.

closed 2026-11-04 by JamBalaya56562