motioneye-project/motioneye #2805: Unable to write stills and movies to another user MNT

Balgroggy, 2023-08-06:
I am fairly new to Raspberry PI and I am using a Raspberry Pi 4B 8Gb. An I/R version of the Raspberry Pi Cam with a USB cam and a USB stick to use as a cache before it gets uploaded to Google Drive via Rclone.

Because of the issues with MotioneyeOS with the new version of the bootloader and other various issues. I've installed the Buster legacy version of Raspbian and installed and tested Rclone working under the default PI account. So the mount point is under

/home/pi/mnt/gdrive/(foldername)

It works well and I've tested the link both ways without an issue. However, after installing Motioneye using the guide here: https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian

And trying to use the Rclone mount point, rebooting Motioneye locks up. I've changed the path temporarily to the USB SDA1 mount and Motioneye works again. Looking at the log, it appears the mnt point under the pi account isn't able to be written. However, changing the permissions all along the chain doesn't allow it to write still. Also after rebooting, the permissions appear to revert.

Is there a way to allow Motioneye to use the Rclone mount point in the PI account, or do I make another account for Motioneye and copy the config for the Motioneye account so it can send to Google drive or another solution that might work?

Thank you for you time and knowledge.

Balgroggy

zagrim, 2023-08-08:
The permissions of a mount point are partly changed at the time of mounting based on the settings in `/etc/fstab` (or given with `mount` command). I'm not familiar with `rclone` but I guess you are using `rclone mount` in some startup script or have configured it so that the config is in `/etc/fstab`? And you've read https://rclone.org/commands/rclone_mount/ and https://rclone.org/drive/ but didn't find any help from there?
Please show the command/config used for mounting, the error you see in the log, and what the permissions on disk are after rebooting.

Balgroggy, 2023-08-13:
Hi Zagrim,
                Sorry its taken so long to reply, I can only work on this at the weekend. I found I was pointing to the wrong folder in the first place. I can run:-

**_rclone sync -v /media/pi/USBDISK/Security/ gdrive:/Security/_**

And its now sending the files, just need to put it in a script and get crontab working with it!

Thanks for the advice, it did help

zagrim, 2023-08-15:
Good to hear you found a solution. Please close this issue if it has been resolved.

Balgroggy, 2023-08-15:
Thanks for the help!

closed 2023-08-15 by Balgroggy