motioneye-project/motioneye #2806: USB storage - permission denied to write

brandflugan, 2023-08-08:
I've recently plugged in a flash storage device to my raspberry pi 3b that I would like to save my recordings in.

I have used both ext4 and fat32 format on my new storage device and changed the root directory for my camera (network camera with pass through) to the mounted path.

The path is /media/piserver/EXTRAHDD/recordings/backyard

However when I try to take a snapshot or waiting for a recording to finish, nothing appears on the storage device.
I checked the logs and found out that I have permission problems:

Aug 08 11:55:21 raspberrypi motion[4541]: [1:ml1:Backyard] [ERR] [EVT] event_image_snapshot: Could not create symbolic link [2023-08-08/11-55
-21.jpg]: Permission denied
.......
raspberrypi meyectl[4948]: PermissionError: [Errno 13] Permission denied: '/media/piserver/EXTRAHDD/recordings/backyard'
.......

I therefor changed the permissions to the EXTRAHDD and all the folders inside to give it read and write access as well as changing the user group to "motion", however that is not helping:

ls -al for the backyard folder gives:
drwxrwxrwx 2 motion motion    4096 Aug  8 11:54 backyard

The disk seems to be mounted properly:
sda           8:0    1 232.4G  0 disk
└─sda1        8:1    1 232.4G  0 part /media/piserver/EXTRAHDD

I have added it to fstab and checked that it mounts with "mount -a":
/dev/sda1    /media/piserver/EXTRAHDD    ext4    defaults    0    0

I even gave read and write access to the motion group itself by:
sudo chmod -R g+rw /media/piserver/EXTRAHDD/recordings

What am I missing here?

**Specs**
motionEye Version | 0.43.0
Motion Version | 4.3.2
OS Version | Raspbian 11

zagrim, 2023-08-10:
When you get "permission denied" on for example for `2023-08-08/11-55-21.jpg`, what are the permissions and ownership of `/media/piserver/EXTRAHDD/recordings/backyard/2023-08-08` AND that file there at that time, without any manual changes?

I'm asking this because changing the read/write permissions of a parent folder doesn't affect the permissions of new files and folders created there. Then again, if motion is able to create a new folder/file in some parent folder, it is indeed weird if it can't symlink that same folder/file.

Alpha-Betty, 2023-08-11:
not sure if related,  but I have an orangePi 5,  i installed motioneye, via the raspberry pi docker image  dontobi/motioneye.rpi.   which is based on motionEye Version | 0.42.1 so probably not related but might as well post my experience on the off chance it helps.  

Anyway I was getting permission issues, and the contain would not even start, I did all the usual things,  gave full recursive permissions to the user,  changed ACL for the user, even tried giving the container full root access, but was still getting permission issue messages in the logs and the container would not start. 

Turned out that the container, needed the directory used to store the data and videos to not exist before creation of the container.   Also if any other container later got access to the motioneye directory then the files became inaccessible to motioneye, and it would be automatically stopped.   maybe it's a security feature i'm unaware of.  

well, only mentioning it because the storage device was also a USB drive

Alpha-Betty, 2023-08-12:
ignore my comment, my issue was related to importing a config another machine,  both arm devices.   But I was importing the config and motioneye in the container would work until the next  reboot , when i would get the message ""/data/motioneye" does not exist or is not writable"  

all working now just had to re import all my cameras and settings from scratch.

JamBalaya56562, 2026-11-12:
The folders below it are open, but `motion` also needs to enter every directory above it. `/media/piserver` is created by udisks for that user only, so `motion` cannot reach `EXTRAHDD` at all; `namei -l /media/piserver/EXTRAHDD/recordings/backyard` shows where it stops. As the drive is in `/etc/fstab` anyway, mounting it outside the user's directory, e.g. at `/mnt/extrahdd`, avoids this. As there was no follow-up since 2023, closing as answered.

closed 2026-11-12 by JamBalaya56562