motioneye-project/motioneye #2808: FYI: Problem with Pillow >10.0.0

TheRaven500, 2023-08-10:
I stumbled over a problem.
There is a problem with the new Pillow version >10.0.0 as described here:
https://stackoverflow.com/questions/76616042/attributeerror-module-pil-image-has-no-attribute-antialias
Option1: Force Pillow 9.5.0 -> pip3 install Pillow==9.5.0
Option2: Change the code.

If i have a free wish: Please make a manual how to install motioneye on debian 12 bookworm. :-)

zagrim, 2023-08-12:
Thanks for reporting and providing a workaround option. The issue was discovered a while ago and there's a PR already pending (#2796).

TheRaven500, 2023-08-12:
Perfect, thank you very much! I spent about 4h to fix this error, so I thought it would be good if others do not have to search as well. ;-)

zagrim, 2023-11-13:
Dev branch should now work ok with wide range of Pillow versions.

closed 2023-11-13 by zagrim