motioneye-project/motioneye #2809: Cannot find mass storage on the web gui

yldzmuhammed, 2023-08-13:
Hello,
I have installed the motioneye on the my local ubuntu server PC based on the suggestions on wiki.
On my PC, I have two different physical disk. One is 256GB Sata SSD, the other is 4 TB regular Sata HDD.
I have format my HDD based on https://help.ubuntu.com/community/InstallingANewHardDrive and I mounth my HDD as /storage/disk1 and when I run the command `df -h` I get the following
`Filesystem      Size  Used Avail Use% Mounted on
tmpfs           783M  1.4M  782M   1% /run
/dev/sda2       233G  8.9G  212G   5% /
tmpfs           3.9G   16K  3.9G   1% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
/dev/sda1       1.1G  6.1M  1.1G   1% /boot/efi
/dev/sdb1       3.6T   40K  3.4T   1% /storage/disk1
tmpfs           783M  4.0K  783M   1% /run/user/1000`
I think I format the disk correctly and mounth it correctly.
But still I can not see the mass storage option on web GUI
 
[image]

What should I do?

zagrim, 2023-08-23:
I'm not aware if there are some specific restrictions with the ME ability to recognise mounted disks and partitions and only filtering out such that the mount point is not accessible to the user ME is running as.
However, you always have the option to use "Custom Path" and enter there the path you want to store all the saved images and videos, which in your case could be `/storage/disk1` (unless you are planning to also use that disk partition to some other purpose, in which case you might want to first create a directory there for MotionEye to keep things more organised - but that's up to you to decide).

JamBalaya56562, 2026-11-12:
Most likely the mount point was not writable by the `motion` user: `/storage/disk1` mounted as in that guide belongs to root, and motionEye used to leave out such disks without a word. Since #3380 (in 0.45.0b1), they are listed with "[No write access]" and cannot be selected until you give `motion` write access, e.g. `sudo chown motion:motion /storage/disk1`. "Custom Path" with that directory works as well, as zagrim said. Closing as fixed by #3380.

closed 2026-11-12 by JamBalaya56562