motioneye-project/motioneye #2810: Unable to open video device

Slipery, 2023-08-16:
Raspberry PI 4
Debian Buster Raspbian 10

motionEye Version | 0.42.1
Motion Version | 4.5.1

Camera RASPBERRY PI NOIR CAMERA BOARD V2

Video device is set to 
Camera Device   /dev/video0
 Camera Type  V4L2 Camera

I have read elsewhere that I should use Local NMAL camera
but when I try that I get the following error "An error occurred.  Refreshing is recommended"
I have been having problems with motion detection so I decided to take a manual snap shot and below is what is seen.  However the 
image appears reasonable on the ME application
[image]
[image]
Any help would be appreciated thanking you.
Cheers Mal

Slipery, 2023-08-17:
If this helps this was the steps I used to install MotionEye

pi Imager enable ssh
Raspberry Pi 4 Buster lite

raspi-config
	Interface options
	 Enable camera
	 Enable i2C
sudo reboot
sudo apt-get update
sudo reboot
vcgencmd get_camera
! displays supported =1 detected = 1, libcamera interfaces =0
raspistill -t 10000 
!do remainder from ssh

sudo su
apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12 -y
curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py
python2 get-pip.py

wget https://github.com/Motion-Project/motion/releases/download/release-4.5.1/pi_buster_motion_4.5.1-1_armhf.deb
dpkg -i pi_buster_motion_4.5.1-1_armhf.deb
apt-get install python-pip python-dev libssl-dev libcurl4-openssl-dev libjpeg-dev libz-dev -y
pip install motioneye
mkdir -p /etc/motioneye
cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
mkdir -p /var/lib/motioneye
cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
systemctl daemon-reload
systemctl enable motioneye
systemctl start motioneye
http://192.168.1.204:8765
add camera
	Local V4L2 Camera
	NMAL Service 16.1

xXAzazelXx, 2023-08-18:
i got the same issue on fresh install

Slipery, 2023-08-19:
Yep I was able to replicate the issued on a Pi Zero W as well

xXAzazelXx, 2023-08-19:
so i don't know if this is the same issue for you guys, but I could see in the logs /var/motioneye/ that I was getting an error that sounded like the camera was already in use.
So I've tried stopping motion service and the motioneye started displaying the camera. So I've just disabled motion service in systemctl  and now it is all working

Solarbytes, 2023-08-26:
I have a similar issue with pi camera 3 noIR, grey preview screen, what I find very confusing on this OS image is that it does not seem to have the most elementary command line commands such as APT or even "shutdown" was not recognized... Is this intentional or did I do something wrong during the install/configuration?

The camera is detected and does take manual snapshots but I can't see the images nor the video streaming (timeout on port 8082 I think it was).

Solarbytes, 2023-08-26:
> so i don't know if this is the same issue for you guys, but I could see in the logs /var/motioneye/ that I was getting an error that sounded like the camera was already in use. So I've tried [...]

How did you stop the service? I tried using systemctl to get a log but the command was not recognized, same as for APT or even "shutdown", and because there is no apt or apt-get, I couldn't install any of the missing software either.

xXAzazelXx, 2023-08-26:
i didn't use the OS image i manually installed motioneye 0.43 on a fresh version of Rasbian lite using this link [https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye](url)

after I've completed all of the steps, and added the camera in, then again I went to the start of the instructions and stopped and disabled motion.

Solarbytes, 2023-08-26:
Pi zero W
Linux raspberrypi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
pi camera 3 wide noIR

thank you for the detailed instructions. I do however face an error:
Package python-pip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python3-pip

I also notice you install python2 but later install python-pip which I interpret (right or wrong I don't know as I am a beginner with Pi & Linux) as python1.

I replaced that with python3-pip and it went ahead with the installation. My doubt at this stage is that the app is probably invoking python and not python3 in some areas, so would everything still work? 

Everything does fire up with some differences in the interface: no update/reboot/shutdown buttons), however I have the same issue as with the installation image: failed to open device, grey screen.

After all was completed, I still have the same issue: the preview window is grey with message cannot open the device, and a still shot capture that exact grey screen.

xXAzazelXx, 2023-08-26:
my recommendation would be to start fresh, flash the latest RaspbianLite to your Pi zero W and follow these:
https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye
I can confirm that those steps plus the stop and disable motion step again at the end, does work with my Pi Zero W.

python-pip is for python2 and as I understand Motioneye only supports python2 not 3. In step 3 of the instruction above you will see that you are installing python2, and pip for python2 as its end of live and does not come installed by default anymore.

EpicLPer, 2023-08-26:
This also seems to happen on the current dev branch (python3), when I reboot my Pi with motion enabled it can't open one camera for some reason, as soon as I disable it and reboot it suddenly works again.

Solarbytes, 2023-08-26:
> my recommendation would be to start fresh, flash the latest RaspbianLite to your Pi zero W and follow these: https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye I can confirm that those steps plus the stop and disable motion step again at the end, does work with my Pi Zero W.
> 
> python-pip is for python2 and as I understand Motioneye only supports python2 not 3. In step 3 of the instruction above you will see that you are installing python2, and pip for python2 as its end of live and does not come installed by default anymore.
> 
> ```
>  curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py
>  python2 get-pip.py
> ```

I got as far as attempting to install pip because python2 wasn't installed, and install python2 failed due to missing dependencies and then there was this recommendation:

Try 'apt --fix-broken install' with no packages (or specify a solution).

but I then got the following error when applying that recommendation:
E: Failed to fetch http://mirror.datamossa.io/raspbian/pool/main/libx/libxinerama/libxinerama1_1.1.4-2_armhf.deb  Unable to connect to mirror.datamossa.io:http:
E: Aborting install.

Solarbytes, 2023-08-26:
> I got as far as attempting to install pip because python2 wasn't installed, and install python2 failed due to missing dependencies and then there was this recommendation:
>
> Try 'apt --fix-broken [...]

well after retrying, it eventually worked, but I now hit a wall with motioneye where I get compilation errors on pillow and the following command does not work:

pi@raspberrypi:~ $ sudo apt-get install python-pil -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python-pil

I installed pillow but the same error persists.

Any ideas?

xXAzazelXx, 2023-08-27:
OK apologies, I had myself confused. I've deleted old posts, this is what I did:

1) Fresh install of Raspbian Lite
2) [Dev 0.43 Motioneye](https://github.com/motioneye-project/motioneye/tree/dev), follow this one instead.
3) On step 3 of `sudo python3 -m pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'` you will run in to compile errors.
I think I installed the following:

```
pip install --upgrade pip setuptools
sudo apt-get install libcurl4-openssl-dev
sudo apt-get install python3-dev
sudo apt-get install libssl-dev
```

4) Run `sudo python3 -m pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'` again, it should work this time.
5) As per the guide, `sudo motioneye_init`
6) At this point you should be able to get to GUI of Motioneye, and detect the camera. Test also with `vcgencmd get_camera`
7) You will get the grey screen even after adding the camera, run `sudo  systemctl stop motion`
If the camera started working, run ` sudo  systemctl disable motion`

Good luck

zagrim, 2023-08-27:
> what I find very confusing on this OS image is that it does not seem to have the most elementary command line commands such as APT or even "shutdown" was not recognized... Is this intentional or did I do something wrong during the install/configuration?

@Solarbytes If you are referring to MotionEyeOS, that is indeed intentional. The image is meant to be a lightweight and compact setup just for running MotionEye on a limited resource device such as the earlier Raspberry Pis or Pi Zeroes, nothing more. For a setup where you want to install other stuff, too, you need to use a Linux distro and install ME separately, such as with the instructions provided by @xXAzazelXx above.

Slipery, 2023-08-27:
Okay have installed on Bullseye Lite I now get a good picture both live and saved when I take a manual snapshot.
 Added a Camera 
        Local V4L2 Camera
         NMAL Service 16.1
It shows up under Video service as 

[image]

But motion detection still does not work.  I have made the settings the same as the other PI Zero which uses the MotioneyOs
[image]

I notice that using the MotionPi OS I am able to set camera device  to 
vc.ril.camera
and Camera Type as MMAL Camera
If I try to select Camera type while adding a camera  i get the same error as I did with Buster.
[image]

and yes I have enabled legacy camera using raspi-config

Solarbytes, 2023-08-27:
> sudo apt-get install libssl-dev

alas, still no success. I did not run into compilation issues but nevertheless installed the additional packages, rebooted and added 3 different setups of camera/codec and I still get the grey screen with message "unable to open video device". If I take a snapshot, I still get that same grey screen on the output. I did disable motion. at this stage, I'm just thinking that the pi camera 3 is not compatible with old software.

Solarbytes, 2023-08-27:
@xXAzazelXx I just remebered that before I could get the Pi camera 3 to work in a GUI terminal window, I had to specify a profile file located under /usr/share/libcamera/ipa/rpi/vc4. Could it be that the same issue is preventing the camera from working in the motioneye GUI?

Also, even if disabling Motion fixes the issue on a Pi camera 2, doesn't that defeat the purpose of having Motioneye? am I correct in assuming that this would prevent the motion detection to work to trigger the camera?

xXAzazelXx, 2023-08-27:
Regarding tuning files, I'm not sure I never had to do that.

Re Motion, my understanding is that Motioneye runs on its own utilizing Motion's files. If Motion is running by itself and using the camera, Motioneye cant get access.
I mean its also part of the official instructions to stop it

danxdz, 2024-05-09:
Hi, no solutions found  after this?

MichaIng, 2024-05-09:
This is no generic error/bug, but it highly depends on your SBC, camera type/device and OS setup. If your case does not match OP exactly: RPi OS Buster on RPi with RPi camera module v2, please open a new issue.

@Slipery I guess also you updated your system in the meantime, using Bullseye or Bookworm, with KMS video stack enabled by default? In this case, one either needs to disable KMS or use libcamerify to work around the fact that motion does not support it. Here is an example systemd unit which does this: https://github.com/motioneye-project/motioneye/issues/2900#issuecomment-1870273642, i.e. the `ExecStart=` line changed. And you need to install libcamerify, of course:
```sh
apt install libcamera-tools
```

danxdz, 2024-05-09:
Hi, thanks a lot, its installing, but yes im installing new cams with , with bullseye, rpi zero w and pi camera IR.

It gives me a grey screen after install, but i get i to work just after stop motion service at systemctl...

MichaIng, 2024-05-09:
Ah yes, the `motion.service` must be disabled, else it blocks access to the video devices for the motionEye controlled motion processes. We must implement this into our init script, so it is disabled automatically: https://github.com/motioneye-project/motioneye/blob/dev/motioneye/extra/linux_init

danxdz, 2024-05-09:
> Ah yes, the `motion.service` must be disabled, else it blocks access to the video devices for the motionEye controlled motion processes. We must implement this into our init script, so it is disabled automatically: https://github.com/motioneye-project/motioneye/blob/dev/motioneye/extra/linux_init
> 
> PR open for this: #2980

Oh thanks, i have strungling with this last days, not found the right info.. but i think at same time motion detection stop work :( motion detection should work right?

MichaIng, 2024-05-09:
Yes, it should. However, this is an entirely different topic then.