motioneye-project/motioneye #2812: Motioneye and Libcamera

Schnitzer50, 2023-08-22:
Hello Motioneyeteam!
How long will it be before Motioneye also supports the new camera drivers for RPi?
Sonnige Grüße aus Ulm, Germany

zagrim, 2023-08-23:
Libcamera support is still "under construction", and that applies to the yet unreleased ME 0.43 with Python3 support. The topic has been discussed here: #2425 (and here: #2683).

As far as I've understood things correctly, there are some workarounds mentioned in #2683, but no complete solution yet. There is a PR (https://github.com/motioneye-project/motioneye/pull/2765) linked to the other issue and it is waiting for testing, with some instructions on how that could be done (but I admit they might not be good instructions in case one is not that tech-savvy).

pingufreak, 2023-11-18:
Working with Debian Bookworm with current patchlevel:

```
apt update 
apt upgrade -y
python3 -m pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'
apt install -y libcamera-v4l2
sed -i 's/\/usr\/local\/bin\/meyectl/\/usr\/bin\/libcamerify \/usr\/local\/bin\/meyectl/' /etc/systemd/system/motioneye.service
systemctl daemon-reload
systemctl enable motioneye --now
```

In motionEye add the first camera. Example:
Local V4L2 Camera
/base/soc/.../.../ov5647...

have a nice day

pingufreak

Martypose, 2023-11-23:
@pingufreak 
im gonna try your solution for bookworm in a pi4 8gb with a rpi cam ov5647
hope it works cause im losing my mind with motioneye haha

Juanderer98, 2023-11-28:
@pingufreak,
Thank you for your suggestions... I (and others) have been struggling with this issue. I tried your suggestion on a fresh install of 64bit raspian bookworm on a RPi 4b 4Gb model.  after trying to execute the third line I get:
...
**error: externally-managed-environment

This environment is externally managed**
...
I assume that this is a new "feature" of Debian Bookworm, that I haven't figured out yet. Is there a way to get motioneye to install directly, or a simple way to set up a venv? 

Thank you for any suggestions...

Juanderer98, 2023-11-28:
@pingufreak
 I tried entering the third line:
    python3 -m pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz' --break-system-packages
in hopes of getting around the venv issue.  It seemed to be working, but then got the following error.  I'm not sure what next to try...

...
Collecting pycurl
  Downloading pycurl-7.45.2.tar.gz (234 kB)
     ------------------------------------- 234.2/234.2 kB 491.7 kB/s eta 0:00:00
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  python setup.py egg_info did not run successfully.
  exit code: 1
  
  [26 lines of output]
  Traceback (most recent call last):
    File "/tmp/pip-install-gp3rsl8m/pycurl_3fffa710a6074f8d9ede04a2314e9dfb/setup.py", line 229, in configure_unix
      p = subprocess.Popen((self.curl_config(), '--version'),
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/usr/lib/python3.11/subprocess.py", line 1024, in __init__
      self._execute_child(args, executable, preexec_fn, close_fds,
    File "/usr/lib/python3.11/subprocess.py", line 1901, in _execute_child
      raise child_exception_type(errno_num, err_msg, err_filename)
  FileNotFoundError: [Errno 2] No such file or directory: 'curl-config'
  
  During handling of the above exception, another exception occurred:
  
  Traceback (most recent call last):
    File "<string>", line 2, in <module>
    File "<pip-setuptools-caller>", line 34, in <module>
    File "/tmp/pip-install-gp3rsl8m/pycurl_3fffa710a6074f8d9ede04a2314e9dfb/setup.py", line 970, in <module>
      ext = get_extension(sys.argv, split_extension_source=split_extension_source)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/tmp/pip-install-gp3rsl8m/pycurl_3fffa710a6074f8d9ede04a2314e9dfb/setup.py", line 634, in get_extension
      ext_config = ExtensionConfiguration(argv)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/tmp/pip-install-gp3rsl8m/pycurl_3fffa710a6074f8d9ede04a2314e9dfb/setup.py", line 93, in __init__
      self.configure()
    File "/tmp/pip-install-gp3rsl8m/pycurl_3fffa710a6074f8d9ede04a2314e9dfb/setup.py", line 234, in configure_unix
      raise ConfigurationError(msg)
  ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory: 'curl-config'
  [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

Encountered error while generating package metadata.

See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
root@raspberrypi:/home/pi#

zagrim, 2023-11-28:
@Juanderer98 you seem to be missing `curl-config`, which should be available in `libcurl4-openssl-dev` package which you need to install. And given that there are a couple of other additional packages for 64-bit RPis in the dev branch installation instructions ([see step 1 here](https://github.com/motioneye-project/motioneye/tree/dev#installation)) you might need to install those, too.

pingufreak, 2023-11-28:
> File "", line 2, in
> File "", line 34, in

Please read the logs / rtfm:

apt install -y curl-config

The other errors: bookworm doesn't come with python2, you really need to
install the dev version of motioneye with pip3.

pingufreak, 2023-11-28:
99,999999% of the time for fixing issues is reading logs / docs and
understanding in detail what's going on. If you just follow this advise,
you can get really good quite fast.

Schnitzer50, 2023-11-28:
Hello everyone, first of all thank you for the detailed information on how to improve MotionEye.For me as a simple user, the instructions are very confusing and I cannot estimate the consequences. Are there no easy update options for Motioneye? I still understand updates via the user interface or terminal instructions for Debian11, and I can also load them from GitHub. Best regards✉️ vom 🖥

Juanderer98, 2023-11-28:
@pingufreak
 Thank you for your guidance.  
-------
_99,999999% of the time for fixing issues is reading logs / docs and
understanding in detail what's going on. If you just follow this advise,
you can get really good quite fast._
-------
I'm still trying to find the applicable log file...  I found '/var/log/apt/term.log' but this doesn't seem to have the information on errors.  By docs, I'm guessing that would be the motioneye wiki, which I'm re-reading.   The most recent error is below:
-------
root@raspberrypi:/home/pi# sed -i 's/\/usr\/local\/bin\/meyectl/\/usr\/bin\/libcamerify \/usr\/local\/bin\/meyectl/' /etc/systemd/system/motioneye.service
sed: can't read /etc/systemd/system/motioneye.service: No such file or directory
-------
Any suggestions are most welcome.

Juanderer98, 2023-11-29:
@pingufreak,
Thank you again for your guidance (and patience).  Maybe I'm getting a little closer to understanding your suggestion to read logs / docs, but probably not.  Anyway, I'm looking through the manual for the sed command.  I'm guessing it is to modify the file motioneye.service in place, by replacing paths but I'm still stuck trying to find the original motioneye.service file.  I tried to do 'find files' with the file manager, but no luck...

ishetkunst, 2024-02-20:
Thanks. This worked!

Schnitzer50, 2024-02-21:
Please explain me how to easily update motioneye with libcamera. Thanks for your help.

Schnitzer50, 2024-02-28:
My installed Motioneye version is 22.4.1. Is there already a newer version? If so, what is the easiest way to install it?

CHARL13is, 2024-03-01:
@pingufreak did you manage to write an article in the end?  I'm trying to use Motioneye with an RPi 5 (Bookworm) with a Pi v2 NOIR camera but having tried all of the various advice and guides I cannot get this to work.  What I could really do with is a step-by-step from a fresh install.  It used to be so easy!

Martypose, 2024-03-02:
> @pingufreak did you manage to write an article in the end? I'm trying to use Motioneye with an RPi 5 (Bookworm) with a Pi v2 NOIR camera but having tried all of the various advice and guides I [...]

I would also appreciate some kind of basic guide for this same setup.

Thank you very much.

birefringence, 2024-06-02:
I have the original v1 camera module connected to a Pi Zero 2 W and the above instructions work in so far as that the module is detected. The picture I see, however, only consists of green stripes. I see some changes when I move the camera. Therefore, I conclude that data is transferred but there seems to be a problem with the pixel format. Everything looks fine directly with libcamera. I tried experimenting with the resolution and motion "palette" setting, but that was unsuccessful.

fikin, 2024-07-05:
> Hello Bello 🐶,
> 
> working with Debian Bookworm with current patchlevel:
> 
> ```
> apt update 
> apt upgrade -y
> python3 -m pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'
> apt install -y libcamera-v4l2
> sed -i 's/\/usr\/local\/bin\/meyectl/\/usr\/bin\/libcamerify \/usr\/local\/bin\/meyectl/' /etc/systemd/system/motioneye.service
> systemctl daemon-reload
> systemctl enable motioneye --now
> ```
> 
> In motionEye add the first camera.
> 
> Example: Local V4L2 Camera /base/soc/.../.../ov5647...
> 
> Please update the wiki, I can't do a pull-request on it 🙈 🙉 🙊:
> 
> [Install on Debian 12 (Bookworm)](https://github.com/motioneye-project/motioneye/wiki/Install-on-Debian-12-(Bookworm))
> 
> Greetings from Trier (Germany)
> 
> 7 Edits 😀
> 
> Have a nice day!
> 
> pingufreak

to confirm these instructions worked for me too.

pi zero 2 w + noir 2 + debian bookworm
following main motioneye installation instructions did not render a working camera.
following https://lookslikematrix.de/raspberry-pi/2023/10/14/motion-bookworm.html rendered working motion but not motioneye.
after applying instruction from above, motioneye started working.

Martypose, 2024-07-08:
Same

Noir v1 IR torch

Bookworm clean install

Pi4 4gb

Working fine

[image]

jinky77, 2024-10-20:
Hi there! Reviving this thread as I'm also trying to install motioneye using @pingufreak's solution.

I stumbled upon `sed: can't read /etc/systemd/system/motioneye.service: No such file or directory` like @Juanderer98 did. I carefully followed the steps above, although I should mention that I did NOT installed motioneye using the official installation guide.

Could any of you guys help me solve this? Thanks!

Martypose, 2024-10-20:
Create manually that file with

sudo nano /etc/systemd/system/motioneye.service

the content is something like 

[Unit]
Description=motionEye Server
After=network.target

[Service]
ExecStartPre=/bin/mkdir -p /run/motioneye
ExecStartPre=/bin/chown root:root /run/motioneye
ExecStart=/usr/bin/libcamerify /homeYOURUSER/motioneye-env/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
User=root
Group=root
Type=simple
Restart=on-failure

[Install]
WantedBy=multi-user.target

-----

The reason you had to create the /etc/systemd/system/motioneye.service file manually is that this file defines how the motionEye service will be managed by systemd, which is the init system used in modern Linux distributions to manage services (including starting, stopping, restarting, and ensuring they run at boot).

Normally, when you install software through package managers (like apt), a service file is automatically created. However, in this case, since you are installing motionEye manually or from a custom method (such as a Python virtual environment), the service file isn't provided by default.

I think.

jinky77, 2024-10-25:
Hey @Martypose thank you very much for taking the time to reply. I ended up not needing to manually set `motion.service` after all, here is what I did:

1. On a brand new Bookworm (Raspberry Pi OS 64-bit) installation, `sudo apt update && apt upgrade -y`. I've used [Raspberry Pi Imager](https://www.raspberrypi.com/software/) to flash the SD card
2. Followed the installation instructions provided [here](https://github.com/motioneye-project/motioneye?tab=readme-ov-file). I just skipped point 2 since `pip` was already installed
3. Installed `libcamera-v4l2` with `apt install -y libcamera-v4l2`
4. Followed the rest of @pingufreak's instructions
5. Rebooted and voilà! The camera was indeed recognized by motionEye

To anyone who might need help with, don't hesitate to write!

[image]

pookguy88, 2024-10-29:
finally got it to work, I Have Raspberry Pi Zero 2 and the Arducam IMX708 - Raspberry Pi Imager and installed the latest Bookworm 64bit lite

followed @pingufreak's instructions

so I used Raspberry Pi Imager and installed the latest Bookworm 64bit lite. Went through everything and booted up Motioneye, couldn't detect my camera.

After reading @jinky77 's comment him saying pip was already installed, I realized I don't think libcamerify was installed on my Bookworm build

So in the `apt install -y libcamera-v4l2` step of the instructions, you need to also have libcamera-tools, so add in:
`sudo apt-get install libcamera-tools`

Because the next step with the sed command is basically a find and replace and it just assumes you have libcamerify installed

So after doing all that and booting up Motioneye, I finally get it to work.

followed @pingufreak's instructions but for the life of me couldn't it get it work.

mark4mike, 2024-11-03:
Great thread, I've been trying to get an Arducam IMX708 Camera Module 3 to work with motioneyeOS for a while on a Pi Zero 2 W.  Thanks to @pingufreak, @jinky77, and @pookguy88!  Just to add to this in case it helps anyone, I had to pretty much do a combo of everyone's steps to some degree and did have to end up installing `pip`.  Additionally, my camera would not work or get detected until doing the following from Arducam's [Wiki:](https://docs.arducam.com/Raspberry-Pi-Camera/Native-camera/12MP-IMX708/#:~:text=Bookworm%20OS%20on%20Pi0%2DPi3)

```
sudo nano /boot/firmware/config.txt

#Find the line: camera_auto_detect=1, update it to:
camera_auto_detect=0

#Find the line: [all], add the following item under it:
dtoverlay=imx708

#Save and reboot.
``` 

After the reboot I was able to add the camera in motioneyeOS.

If it helps anyone, here were the steps I followed to get mine working:

1. Fresh install of Raspberry Pi OS Lite (64-bit) Bookworm
2. Update OS
```
sudo apt update
sudo apt upgrade -y
sudo reboot
``` 
3. [Install motionEyeOS](https://github.com/motioneye-project/motioneye?tab=readme-ov-file#:~:text=on%20Weblate.-,Installation,-These%20install%20instructions). I used @pingufreak's method but had to do a couple pre-requisites to get `pip` installed from the link @jinky77 provided, following the steps based on my architecture. 64-Bit Bookworm in my case:
```
sudo apt update
sudo apt --no-install-recommends install ca-certificates curl python3 python3-dev libcurl4-openssl-dev gcc libssl-dev
grep -q '\[global\]' /etc/pip.conf 2> /dev/null || printf '%b' '[global]\n' | sudo tee -a /etc/pip.conf > /dev/null
sudo sed -i '/^\[global\]/a\break-system-packages=true' /etc/pip.conf 
curl -sSfO 'https://bootstrap.pypa.io/get-pip.py'
sudo python3 get-pip.py
rm get-pip.py
sudo python3 -m pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'
``` 
4. I then followed @pookguy88's info and installed `libcamera-v412` and `libcamera-tools`
```
sudo apt install -y libcamera-v4l2
sudo apt install -y libcamera-tools
```
5. Then initiated motionEyeOS:
```
sudo motioneye_init
sudo sed -i 's/\/usr\/local\/bin\/meyectl/\/usr\/bin\/libcamerify \/usr\/local\/bin\/meyectl/' /etc/systemd/system/motioneye.service
sudo systemctl daemon-reload
sudo systemctl enable motioneye --now
```  
6. This was when I had edited `/boot/firmware/config.txt` as I mentioned above and then rebooted.
7. Navigate to https://[raspberry pi IP address]:8765

After the last reboot I was able to add my camera like normal.  Of course `libcamera-v412`, and `libcamera-tools` packages can be installed before installing motionEyeOS but this was just the order I did it in.  

Thanks all!

zagrim, 2024-11-04:
Thanks for all so far for sharing your solutions! :+1: :+1: :+1: 
One small clarification @mark4mike though to avoid confusion in others reading this: MotionEyeOS is a stripped-down Linux distro by itself, and what you were installing is the standalone MotionEye application :smiley:

mark4mike, 2024-11-04:
> Thanks for all so far for sharing your solutions! 👍 👍 👍 One small clarification @mark4mike though to avoid confusion in others reading this: MotionEyeOS is a stripped-down Linux distro by itself, and what you were installing is the standalone MotionEye application 😃

That's right, my mistake.  The previous times I had played around with it I had always just flashed SD cards directly with motioneyeOS rather than installing it on Raspberry Pi OS.  Thanks for clarifying!

Matthew1471, 2024-11-28:
This thread has useful info spread across it and each of the provided advice is missing some commands.. Taking a microSD and putting Bookworm Lite 64 bit on it, using the imager customisation to give it the WiFi password and enable SSH and then booting and running:

```
# All future commands require root
sudo -i

# Update distro
apt-get update
apt-get dist-upgrade

# Install python3-pip
apt-get install python3-pip

# Fix motioneye dependencies
apt-get install python3-jinja2 python3-pil python3-pycurl python3-babel python3-boto3 python3-tornado

# Override pip limitation
grep -q '\[global\]' /etc/pip.conf 2> /dev/null || printf '%b' '[global]\n' | tee -a /etc/pip.conf > /dev/null
sed -i '/^\[global\]/a\break-system-packages=true' /etc/pip.conf

# Install motioneye
pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'
motioneye_init

# RPI: Fix camera on Bookworm
apt install -y libcamera-v4l2 libcamera-tools
sed -i 's/\/usr\/local\/bin\/meyectl/\/usr\/bin\/libcamerify \/usr\/local\/bin\/meyectl/' /etc/systemd/system/motioneye.service
systemctl daemon-reload
systemctl restart motioneye.service
```

I also like:

```
# RPI: Force 5GHz.
nmcli connection modify preconfigured 802-11-wireless.band a

# Install fail2ban, iptables-persistent, unattended-upgrades
apt-get install fail2ban iptables-persistent unattended-upgrades

# NAT MotionEye to port 80 using iptables.
#-A PREROUTING -p tcp -m tcp --dport 80 -j REDIRECT --to-ports 8765

# nano /etc/fail2ban/jail.conf so backend=systemd

# Clear logs
journalctl --rotate
journalctl --vacuum-time=1s
```

Hope this saves someone a lot of time!

mkalle333, 2024-12-25:
@Matthew1471

Thank you so much for the summary, it worked flawlessly on my Raspberry Model 4 B with Camera Module 3.
Might I suggest adding
`#sudo systemctl enable motioneye`
to start motioneye automatically on start/reboot

Or would it have done anyways? I am mot sure.

Matthew1471, 2025-01-22:
@mkalle333 you're welcome... no need to add enable.. the install script already does that.

I had issues with Wi-Fi on all my Raspberry Pi 4s due to a buggy driver (brcmfmac).. so I've now added this too:

# RPI: Fix Wi-Fi (https://github.com/raspberrypi/linux/issues/6049#issuecomment-2595167933)
# either below or append brcmfmac.feature_disable=0x282000 to /boot/firmware/cmdline.txt
echo "options brcmfmac feature_disable=0x282000" > /etc/modprobe.d/brcmfmac.conf

delanthear, 2025-01-31:
> This thread has useful info spread across it and each of the provided advice is missing some commands.. Taking a microSD and putting Bookworm Lite 64 bit on it, using the imager customisation to give it the WiFi password and enable SSH and then booting and running:
> 

Gah!

This all worked fine for me last week, amazing job pulling it together.  

However though, after a reinstall (using Raspberry Pi OS Lite 64 bit this time rather than the full desktop install, could that be it?) I cannot get MotionEye to detect the camera.  Any idea of where to look for clues?  The camera is working as I just took a shot with it rpicam-stilll.

Matthew1471, 2025-01-31:
> Gah!
>
> This all worked fine for me last week, amazing job pulling it together.
>
> However though, after a reinstall (using Raspberry Pi OS Lite 64 bit this time rather than the full desktop install, [...]

I had issues sometimes installing packages due to a server error (HTTP 400).. so I'd check each command succeeded. Did you try a reboot after (so it's not in use with rpicam-still etc)? I use only Lite so it definitely works with Lite.. make sure you didn't select Legacy Lite?

delanthear, 2025-02-01:
Yeah, I've rerun installed them multiple times and I can't find any errors
anywhere!  I'm going to reimage it with a desktop later and just try again
later.

delanthear, 2025-02-01:
Yeah, a re-image with the desktop version and it still not working.  I followed these instructions exactly, with the same Pi and Camera before and it was fine, but not now.

Just looking at the output from the above commands, is any of this interesting?

After the apt-get -y dist-upgrade
```
The following packages were automatically installed and are no longer required:
  libcamera0.3 libwlroots12
```

and looks like it replaces libcamera0.3 with 0.4
```
The following NEW packages will be installed:
  libcamera0.4 libwlroots-0.18 libxcb-errors0 linux-headers-6.6.74+rpt-common-rpi linux-headers-6.6.74+rpt-rpi-2712 linux-headers-6.6.74+rpt-rpi-v8 linux-image-6.6.74+rpt-rpi-2712
  linux-image-6.6.74+rpt-rpi-v8 linux-kbuild-6.6.74+rpt rasputin
```

this:  (which I suspect has nothing to do with it)
```
dpkg: lxinput: dependency problems, but removing anyway as you requested:
 lxde depends on rasputin | lxinput; however:
  Package rasputin is not installed.
  Package lxinput is to be removed.

```

I can see the camera here fine:
`ls -l /dev/ | grep video
crw-rw----  1 root  video 239,   0 Feb  1 13:49 cec0
crw-rw----+ 1 root  video 237,   0 Feb  1 13:49 media0
crw-rw----+ 1 root  video 237,   1 Feb  1 13:49 media1
crw-rw----+ 1 root  video 237,   2 Feb  1 13:49 media2
crw-rw----+ 1 root  video 237,   3 Feb  1 13:49 media3
crw-rw----+ 1 root  video  81,  14 Feb  1 13:49 v4l-subdev0
crw-rw----  1 root  video  10, 124 Feb  1 13:49 vchiq
crw-rw----  1 root  video  10, 125 Feb  1 13:49 vcio
crw-rw----  1 root  video  10, 122 Feb  1 13:49 vcsm-cma
crw-rw----+ 1 root  video  81,  13 Feb  1 13:49 video0
crw-rw----+ 1 root  video  81,   7 Feb  1 13:49 video10
crw-rw----+ 1 root  video  81,   9 Feb  1 13:49 video11
crw-rw----+ 1 root  video  81,  10 Feb  1 13:49 video12
crw-rw----+ 1 root  video  81,   0 Feb  1 13:49 video13
crw-rw----+ 1 root  video  81,   1 Feb  1 13:49 video14
crw-rw----+ 1 root  video  81,   2 Feb  1 13:49 video15
crw-rw----+ 1 root  video  81,   3 Feb  1 13:49 video16
crw-rw----+ 1 root  video  81,  11 Feb  1 13:49 video18
crw-rw----+ 1 root  video  81,   4 Feb  1 13:49 video20
crw-rw----+ 1 root  video  81,   5 Feb  1 13:49 video21
crw-rw----+ 1 root  video  81,   6 Feb  1 13:49 video22
crw-rw----+ 1 root  video  81,   8 Feb  1 13:49 video23
crw-rw----+ 1 root  video  81,  12 Feb  1 13:49 video31
`

and the camera is working:

` rpicam-still
[0:03:42.433416410] [1671] ERROR IPAModule ipa_module.cpp:171 Symbol ipaModuleInfo not found
[0:03:42.434173021] [1671] ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info
[0:03:42.434284167] [1671]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+50-83cb8101
[0:03:42.576803849] [1674]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:03:42.582547524] [1674]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media1
[0:03:42.582775023] [1674]  INFO RPI pipeline_base.cpp:1121 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Preview window unavailable
Mode selection for 1296:972:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 3296
    SGBRG10_CSI2P,1296x972/0 - Score: 1000
    SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1567
Stream configuration adjusted
[0:03:42.598697355] [1671]  INFO Camera camera.cpp:1202 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-SGBRG10_CSI2P
[0:03:42.599606570] [1674]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
#0 (0.00 fps) exp 33239.00 ag 8.00 dg 1.00
#1 (30.01 fps) exp 33239.00 ag 8.00 dg 1.00
etc
`

From what I can tell, the service config is set up to use libcamerify correctly:
`/etc/systemd/system $ cat motioneye.service 
[Unit]
Description=motionEye Server
After=network.target local-fs.target remote-fs.target

[Service]
User=motion
RuntimeDirectory=motioneye
LogsDirectory=motioneye
StateDirectory=motioneye
ExecStart=/usr/bin/libcamerify /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
Restart=on-abort

[Install]
WantedBy=multi-user.target
`

Perhaps I damaged the camera somehow between when it was working before and now, but then why would I be able to take stills with it?

I'm far too clueless with linux to actually debug what's going on here, but from what I can see I can't find any errors or clues as to why it wouldn't be working.  The dropdown just says (no cameras)

Matthew1471, 2025-02-01:
@delanthear okay, I'm able to reproduce this.. so last night after your message I noticed there was some pending packages to update for libcamera (I auto upgrade every evening).. I let them update last night.. rebooted the machine today and I also have the no camera message on a previously working install.

```
2025-02-01 00:48:57,339 DEBUG pkgs that look like they should be upgraded: libcamera-ipa
libcamera-tools
libcamera-v4l2
libpisp-common
libpisp1
linux-headers-rpi-2712
linux-headers-rpi-v8
linux-image-rpi-2712
linux-image-rpi-v8
linux-libc-dev
rpicam-apps-lite
...
2025-02-01 06:16:52,146 INFO Packages that will be upgraded: libcamera-ipa libcamera-tools libcamera-v4l2 libpisp-common libpisp1 linux-headers-rpi-2712 linux-headers-rpi-v8 linux-image-rpi-2712 linux-image-rpi-v8 linux-libc-dev rpicam-apps-lite
2025-02-01 06:16:52,146 INFO Writing dpkg log to /var/log/unattended-upgrades/unattended-upgrades-dpkg.log
2025-02-01 06:21:01,118 INFO All upgrades installed
2025-02-01 06:21:48,967 INFO Packages that were successfully auto-removed: libcamera0.3 linux-headers-6.6.51+rpt-common-rpi linux-headers-6.6.51+rpt-rpi-2712 linux-headers-6.6.51+rpt-rpi-v8 linux-image-6.6.51+rpt-rpi-2712 linux-image-6.6.51+rpt-rpi-v8 linux-kbuild-6.6.51+rpt
```

One of these being upgraded has broken motioneye... for existing and new installs.

I'll do some more digging.. a work-around will be to find which package broke it and re-install the older version.

delanthear, 2025-02-01:
Works! :D

Matthew1471, 2025-02-06:
More people are having issues with libcamera0.4 and motion and there's now more details here:

https://github.com/raspberrypi/libcamera/issues/235
https://github.com/raspberrypi/libcamera/issues/220
https://bugs.libcamera.org/show_bug.cgi?id=253

bit-bash, 2025-02-16:
Using an RPi 5 with RP Camera V3.
Had some issues after following: https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2506840345
Couldn't get the camera to work.
Turns out I actually have to do this line:
# apt-get -y install libcamera-v4l2 libcamera-tools
That is, I uncommented it then my camera works. In the original post, I thought this line was commented out because it was part of the libcamera downgrading process.
I've got the very latest bookworm updates, so I didn't have to downgrade libcamera0.4 - it's now working.

Matthew1471, 2025-02-17:
> Using an RPi 5 with RP Camera V3 with bookworm 64bit lite. Had some issues after following: [#2812 (comment)](https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2506840345) Couldn't get the camera to work. Turns out I actually have to do this line: `# apt-get -y install libcamera-v4l2 libcamera-tools` That is, I uncommented it then my camera works. In the original post, I thought this line was commented out because it was part of the libcamera downgrading process. I've got the very latest bookworm updates, so I didn't have to downgrade libcamera0.4 - it's now working.

Yes the workaround is no longer required as they fixed it in the original repo : https://github.com/raspberrypi/libcamera/issues/235#issuecomment-2647863551

delanthear, 2025-02-17:
> repo : raspberrypi/libcamera#235 (comment)

Ah great.  Is there an easy way to regress the workaround?

Rudios81, 2025-02-21:
I have been trying and debugging the Motioneye over the last 2 days, started to give up just when I stumbled on this thread.
In the end, I had to alter the motioneye.service file as indicated by @pingufreak making use of libcamerify
For some reason, it did not work when using the motion_binary declaration in the motioneye.conf file, neihter did the trick from some other thread, creating a motion.sh in my user home folder trying to start motion using libcamerify.

So altering the motioneye.service file by including libcamerify made my setu work!
RaspberryPI 4B+ 
RaspberryPI cam 3

Matthew1471, 2025-02-21:
> I have been trying and debugging the Motioneye over the last 2 days, started to give up just when I stumbled on this thread. In the end, I had to alter the motioneye.service file as indicated by [...]

That's what ```sed -i 's/\/usr\/local\/bin\/meyectl/\/usr\/bin\/libcamerify \/usr\/local\/bin\/meyectl/' /etc/systemd/system/motioneye.service``` does in my instructions.. you may wish to check the other tweaks as there's a few other things to get things like Wi-Fi stable: https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2506840345

Rudios81, 2025-02-21:
> > I have been trying and debugging the Motioneye over the last 2 days, started to give up just when I stumbled on this thread. In the end, I had to alter the motioneye.service file as indicated by [@pingufreak](https://github.com/pingufreak) making use of libcamerify For some reason, it did not work when using the motion_binary declaration in the motioneye.conf file, neihter did the trick from some other thread, creating a motion.sh in my user home folder trying to start motion using libcamerify.
> > So altering the motioneye.service file by including libcamerify made my setu work! RaspberryPI 4B+ RaspberryPI cam 3
> 
> That's what `sed -i 's/\/usr\/local\/bin\/meyectl/\/usr\/bin\/libcamerify \/usr\/local\/bin\/meyectl/' /etc/systemd/system/motioneye.service` does in my instructions.. you may wish to check the other tweaks as there's a few other things to get things like Wi-Fi stable: [#2812 (comment)](https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2506840345)

That are indeed the changes I made.
I am working with a wired RPI4, so no wireless issues there.
On the other hand, been banging my head for 2 days now to get my new PI Zero 2W connected without any lick, maybe these tweaks are helpful for that, I'll take a look, thanks for suggesting.

Matthew1471, 2025-02-21:
> Ah great.  Is there an easy way to regress the workaround?
> […](#)
> 
>  Using an RPi 5 with RP Camera V3 with bookworm 64bit lite. Had some issues
>  after following: [#2812](https://github.com/motioneye-project/motioneye/issues/2812) (comment)
>  <[#2812 (comment)](https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2506840345)>
>  Couldn't get the camera to work. Turns out I actually have to do this line: #
>  apt-get -y install libcamera-v4l2 libcamera-tools That is, I uncommented
>  it then my camera works. In the original post, I thought this line was
>  commented out because it was part of the libcamera downgrading process.
>  I've got the very latest bookworm updates, so I didn't have to downgrade
>  libcamera0.4 - it's now working.
> 
>  Yes the workaround is no longer required as they fixed it in the original
>  repo : [raspberrypi/libcamera#235](https://github.com/raspberrypi/libcamera/issues/235) (comment)
>  <[raspberrypi/libcamera#235 (comment)](https://github.com/raspberrypi/libcamera/issues/235#issuecomment-2647863551)>
> 
>  <[#2812 (comment)](https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2663891934)>,
>  ([motioneye-project/motioneye#2812](https://github.com/motioneye-project/motioneye/issues/2812))
>  <[#2812 (comment)](https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2663891934)>
> 
>  Using an RPi 5 with RP Camera V3 with bookworm 64bit lite. Had some issues
>  after following: [#2812](https://github.com/motioneye-project/motioneye/issues/2812) (comment)
>  <[#2812 (comment)](https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2506840345)>
>  Couldn't get the camera to work. Turns out I actually have to do this line: #
>  apt-get -y install libcamera-v4l2 libcamera-tools That is, I uncommented
>  it then my camera works. In the original post, I thought this line was
>  commented out because it was part of the libcamera downgrading process.
>  I've got the very latest bookworm updates, so I didn't have to downgrade
>  libcamera0.4 - it's now working.
> 
>  Yes the workaround is no longer required as they fixed it in the original
>  repo : [raspberrypi/libcamera#235](https://github.com/raspberrypi/libcamera/issues/235) (comment)
>  <[raspberrypi/libcamera#235 (comment)](https://github.com/raspberrypi/libcamera/issues/235#issuecomment-2647863551)>
> 
>  <[#2812 (comment)](https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2663891934)>,

Sorry for not replying sooner..

```apt-mark unhold libpisp-common```

Then assuming you'd like to update to all the new things

```apt-get update
apt-get dist-upgrade```

DifferentComputers, 2025-05-30:
> That's what `sed -i 's/\/usr\/local\/bin\/meyectl/\/usr\/bin\/libcamerify \/usr\/local\/bin\/meyectl/' /etc/systemd/system/motioneye.service` does in my instructions.. you may wish to check the [...]

Just want to chime in here to say that after literally days of trying to get an ArduCam 8mp IMX219 working with MotionEye on Raspberry Pi OS lite, this command finally did the trick. Thank you!

Just for the search engines:
Linux Dash 6.12.25+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux

[0:12:48.209860306] [2125]  INFO Camera camera.cpp:1205 configuring streams: (0) 1640x1232-YUV420 (1) 1640x1232-SBGGR10_CSI2P
[0:12:48.210591034] [2128]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Sel

MichaIng, 2026-11-13:
#3546 added a native light implementation for this: Set `libcamera_v4l2 /usr/libexec/<host-triple>/libcamera/v4l2-compat.so` (provided by the `libcamera-v4l2` package) in your `motioneye.conf` to have it internally applied for detecting and accessing RPi camera modules. `libcamerify` does exactly the same, but the `libcamera-tools` package adds a bunch of unnecessary dependencies.

Running `motioneye_init` will ask you whether it shall be setup when detecting an RPi, and can be also used on existing installs to save the manual `motioneye.conf` edit.

closed 2026-11-13 by MichaIng