motioneye-project/motioneye #2814: Unable to see the stars and no way to adjust the exposure!

QuickyTank, 2023-09-03:
Hello,
I built a remote camera using motioneyeOS and raspicam V2.0 of both regular and Noir cameras. 
Unfortunately I can't see any of the stars I can see with naked eye. 
How can I fix this? Or is it fixable? 
Why is there no exposure or any adjustments to the way the camera works? All I get to chose is the resolution. 
Thanks for any help you can provide me to get up and running.

acyanic, 2023-11-08:
I would suggest that the camera adjustments will need to be done in the raspicam parameters.  

This might help [ZivaVatra/raspberry-pi-astro-capture](https://github.com/ZivaVatra/raspberrypi-astro-capture)

Also note that raspicam is deprecated and the new camera stack is [libcamera](https://www.raspberrypi.com/documentation/computers/camera_software.html#introducing-the-raspberry-pi-cameras).

JamBalaya56562, 2026-11-12:
As acyanic said, exposure is a setting of the camera driver, not of motionEye. With a V4L2 camera, any controls the driver offers show up under Video Device, and anything else can be passed to Motion through "Extra Motion Options"; long exposures for stars are beyond what Motion's video capture is made for, so a dedicated tool like the one linked above fits better. Closing as answered.

closed 2026-11-12 by JamBalaya56562