motioneye-project/motioneye #2816: Camera reconnected, videos are generated but the dashboard shows "Unable to open video device"

swooboo, 2023-09-05:
Didn't find a similar issue so this might be a new one. I've manually disconnected my USB and then reconnected after a couple of hours. After that, the dashboard shows grey image as in the screenshot below. However, images and videos are generated as usual, in the screenshot below there are videos that appear after the disconnect, and they are proper videos. Not only that - when the motion is detected and a video is starting to be recorded, the grey screen gets the red border that indicates the start of a recording, but the resulting videos are normal and not grey screen, as mentioned before.

Unable to detect since 14:16:
[image]

But there are videos from just now:
[image]

Using Raspbian GNU/Linux 11 (bullseye):
```
root@raspberrypi:~# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
root@raspberrypi:~#
```
Logs:
```
root@raspberrypi:~# service motioneye status
● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-08-28 21:41:38 IDT; 1 weeks 0 days ago
   Main PID: 347 (meyectl)
      Tasks: 10 (limit: 1472)
        CPU: 2d 23h 52min 14.218s
     CGroup: /system.slice/motioneye.service
             ├─  347 /usr/bin/python2.7 /home/swooboo/.local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
             ├─  743 /usr/bin/python2.7 /home/swooboo/.local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
             └─25192 /usr/bin/motion -n -c /etc/motioneye/motion.conf -d 5

Sep 05 19:34:16 raspberrypi meyectl[347]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
Sep 05 19:34:26 raspberrypi meyectl[347]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
Sep 05 19:34:36 raspberrypi meyectl[347]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
Sep 05 19:34:46 raspberrypi meyectl[347]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
Sep 05 19:34:56 raspberrypi meyectl[347]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
Sep 05 19:35:06 raspberrypi meyectl[347]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
Sep 05 19:35:16 raspberrypi meyectl[347]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
Sep 05 19:35:26 raspberrypi meyectl[347]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
Sep 05 19:35:36 raspberrypi meyectl[347]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
Sep 05 19:35:46 raspberrypi meyectl[347]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
root@raspberrypi:~#
```
```
root@raspberrypi:~# tail /var/log/motion.log
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[1:ml1:Camera1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/v4l/by-id/usb-Hewlett_Packard_HP_Webcam_HD_4310-video-index0: No such file or directory
[1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
[1:ml1:Camera1] [ALR] [VID] v4l2_device_open: Failed to open video device /dev/v4l/by-id/usb-Hewlett_Packard_HP_Webcam_HD_4310-video-index0: No such file or directory
[1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
root@raspberrypi:~#
```
Camera:
```
root@raspberrypi:~# v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/video31
        /dev/media0

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/video20
        /dev/video21
        /dev/video22
        /dev/video23
        /dev/media1
        /dev/media2

HP Webcam HD 4310: HP Webcam HD (usb-3f980000.usb-1.4):
        /dev/video0
        /dev/video1
        /dev/media3

root@raspberrypi:~#
```

I have disabled the service `motion` several days ago, I might have installed it before motioneye and it looked like I had two motion processes running, one belonging to motioneye and the other not. But if it is a service that's required by the motioneye service then that is what might've messed it up. Here's the status of the service:

```
root@raspberrypi:~# service motion status
● motion.service - Motion detection video capture daemon
     Loaded: loaded (/lib/systemd/system/motion.service; disabled; vendor preset: enabled)
     Active: inactive (dead)
       Docs: man:motion(1)
root@raspberrypi:~#
```

swooboo, 2023-09-05:
Disconnecting and reconnecting the camera didn't help. Restarted the motioneye service (`sudo service motioneye stop` and then start) fixed it.

MichaIng, 2024-05-09:
PR open to disable the `motion.service` automatically: #2980

Closing this, since you fixed it for yourself already.

closed 2024-05-09 by MichaIng