motioneye-project/motioneye #2818: Webhook does not seem to be executed despite motion detection.

hajue54, 2023-09-08:
Hello everyone.
I would like to use motioneye as a motion detector for my lamp control (tasmota). The command is executed correctly in the web-ui, but there is no reaction in the (possible) output via webhook. This problem has already been discussed, but it seems that the solution has not been done. What to do?

zagrim, 2023-09-17:
If you are referring to either #2704 or #2573, opening yet another issue won't do any good. What you might do to improve things is to verify with some manual testing (with `curl` for example) if the service you call with webhook really is requiring some additional HTTP headers, as discussed in #2573, and post the results there (to keep things organised).
If you are referring to something else, please provide more details.

hajue54, 2023-09-20:
Hi Esa,

I have found the communication problem. The direct communication to Tasmota will be ignored from motioneye - with curl too. The communication with shelly works fine (web hook and command curl)! My workaround for controlingTasmota is using a shell script. 
(I can’t find out the problem between motioneye and Tasmota )

Best regards

Hans-Juergen Geissler

zagrim, 2023-09-26:
What does that shell script then use to communicate with Tasmota API? Something else than web requests? Or does it add some "magic" things to the request (headers, parameters etc)?

And what did you refer to with this anyway:
> This problem has already been discussed, but it seems that the solution has not been done.

hajue54, 2023-09-26:
Hi Esa,

Here is my shell-script:

#!/bin/bash 
curl http://192.168.20.28/light/0?turn=on
curl http://192.168.20.31/cm?cmnd=power1%20on

This is for both systems.

Here the responds:

Tasmota:

{"POWER":"ON"}

Shelly-Dimmer:

{"ison":true,"source":"http","has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"mode":"white","brightness":100,"transition":0}

Command shelly1:  curl http://192.168.20.144/relay/0?turn=on

zagrim, 2023-09-27:
Ok, so `curl http://192.168.20.31/cm?cmnd=power1%20on` is for Tasmota, am I correct? I think the issue is that there is a space (" ", URL encoded "%20") in there, as there was this other issue #1916 where a user reported having identical issue, and even that they found a solution, but I just don't know whether it was running the current dev branch of MotionEye (instead any of the official releases which are pretty old by now) or if it was something else. And thus I just asked about that, I hope we'll get a response.
In the meantime: which version of MotionEye are you running?

closed 2026-11-13 by JamBalaya56562