motioneye-project/motioneye #2826: Make stream template editable / change to fullsize

Bluscream, 2023-09-21:
I am biting my teeth out at this for a while. i want to embed my cams in a few different scenarios (1 big page per cam/grid cams) but i always come into the issue that the size of the stream is different and hardcoded, so everything just results in either tiny images on a large viewport or so big that it's overflowing. ([Context](https://www.phind.com/agent?cache=clmt24ic00004l608opmx4hiq))

## Single example:
What i expect:
[image]

What i get:
[image]

## Grid example:

What i expect:
[image]

What i get:
[image]

zagrim, 2023-09-23:
It's not quite clear to whether you are asking for a change in MotionEye, or help with some personal project? I mean, if ME UI gives you what you wish (I'm referring to the grid example here), why not just use it? And if you just want to have a view-only type of grid view, copying enough of the ME UI code should do the trick. However, it is rather tricky to get things scaled up just enough, and it requires taking into account the frame size and aspect ratio of each of the cameras, see the [updateLayout function](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/static/js/main.js#L1102). Of course you wouldn't need all those calculations if you don't have any surrounding components for which space should be spared, and only a static layout.

Bluscream, 2023-09-23:
I just want an easy way to embed a stream/few streams without having to re-write the whole motioneye UI for that. Can't you make the streams have properties to restrict them to their container's size? Phind says i can't restrict them from outside their embed. Ideally you would make the stream HTML editable so i can tell the <image> tag to stay at 100% size

zagrim, 2023-09-24:
The streaming URL is served directly by Motion, so we can't do anything with its content. See [Motions docs for Live Stream](https://motion-project.github.io/motion_config.html#OptDetail_Stream). Motion seems to support putting multiple camera streams in one page (by specifying `stream_port` not on camera level but on top level), but since MotionEye manages the Motion config for you and doesn't really support that sort of thing I suspect that it is not possible to hack that to work together with MotionEye.

However we do also have Embed URL for each camera, produced by MotionEye, which is meant just for this purpose. Have you tried that out? It does require authentication, though, which might (or might not) be an issue) for your intended purpose.

Bluscream, 2023-09-25:
> The streaming URL is served directly by Motion, so we can't do anything with its content. See [Motions docs for Live Stream](https://motion-project.github.io/motion_config.html#OptDetail_Stream). [...]

Wallpaper engine does not support so only basic auth (user:pw@site) would work

JamBalaya56562, 2026-11-15:
@Bluscream The camera's Embed URL (`http://<host>:8765/picture/<id>/frame/`) already fits the picture to the window it is shown in, by width or height depending on the aspect ratio, so it should give the layout you expect in both examples. What's missing is the login, which Wallpaper Engine can't type in. #3524 (open) adds API keys, sent as `Authorization: Bearer <key>` or as the basic-auth password. Can Wallpaper Engine send one of those?