motioneye-project/motioneye #2828: Dropbox - Service error

lucadjc, 2023-09-25:
Hello,
i'm configuring dropbox backup and I'm getting after the key handshaking the following error

ERROR: dropbox: request failed: POST data should be bytes, an iterable of bytes, or a file object. It cannot be of type str.

guaycuru, 2024-01-18:
I'm having the same issue on [0.43.1b1](https://github.com/motioneye-project/motioneye/releases/tag/0.43.1b1)

tapioko, 2024-07-25:
I'm having the same issue on [0.43.1b2](https://github.com/motioneye-project/motioneye/releases/tag/0.43.1b2).

Changing `body = json.dumps(body)` to `body = json.dumps(body).encode()` in `Dropbox#test_access` (_uploadservices.py_) seems to fix it. The output on UI is "Accessing the upload service succeeded![object Object]".

Somewhat related: https://github.com/motioneye-project/motioneye/issues/2408

JamBalaya56562, 2026-10-25:
Fixed by #3376 (merged into `dev` on 2026-06-28): `Dropbox.test_access` now sends the JSON body as bytes, which is the `.encode()` fix suggested earlier in this thread. It will be included in v0.45.0. Until then, you can test it by installing from `dev`.

Closing as completed. Thanks to everyone who reported and confirmed this.

closed 2026-10-25 by JamBalaya56562