motioneye-project/motioneye #2831: Raspberry Pi Zero 2 W & Camera Module 3, should it work?

D-an-W, 2023-09-28:
Hi folks, I have managed to get MotionEye installed and the camera working but can't get any videos to record.

Images are captured ok and it looks to be detecting motion ok but so far not had any recordings created.

I am using the dev branch of MotionEye and the [workaround](https://github.com/motioneye-project/motioneye/issues/2683#issuecomment-1399310407) to enable the Camera Module 3

D-an-W, 2023-10-04:
This is the log from boot to the first motion detection which are the last three lines...

```Oct 04 14:25:29 raspberrypi systemd[1]: Started motionEye Server.
Oct 04 14:25:36 raspberrypi meyectl[473]: configure_logging cmd motioneye: False
Oct 04 14:25:36 raspberrypi meyectl[473]: configure logging to file: None
Oct 04 14:25:36 raspberrypi meyectl[473]:     INFO: hello! this is motionEye server 0.43.0
Oct 04 14:25:39 raspberrypi motion[618]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Oct 04 14:25:39 raspberrypi motion[618]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Oct 04 14:25:39 raspberrypi motion[618]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Oct 04 14:25:39 raspberrypi motion[618]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.5.1 Started
Oct 04 14:25:39 raspberrypi motion[618]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Oct 04 14:25:39 raspberrypi motion[618]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Oct 04 14:25:40 raspberrypi motion[618]: [1:ml1:Camera Modu] [ERR] [VID] v4l2_fps_set: Error setting fps. Return code -1
Oct 04 14:25:41 raspberrypi meyectl[473]:     INFO: cleanup started
Oct 04 14:25:41 raspberrypi meyectl[473]:     INFO: wsswitch started
Oct 04 14:25:41 raspberrypi meyectl[473]:     INFO: tasks started
Oct 04 14:25:41 raspberrypi meyectl[473]:     INFO: mjpg customer garbage collector has started
Oct 04 14:25:41 raspberrypi meyectl[473]:     INFO: server started
Oct 04 14:30:23 raspberrypi motion[618]: [1:ml1:Camera Modu] [ERR] [ENC] ffmpeg_set_codec: Could not open codec Cannot allocate memory
Oct 04 14:30:23 raspberrypi motion[618]: [1:ml1:Camera Modu] [ERR] [ENC] ffmpeg_open: Failed to allocate codec!
Oct 04 14:30:23 raspberrypi motion[618]: [1:ml1:Camera Modu] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for movie output.```

D-an-W, 2023-10-04:
Solved by changing video type and carrying out a reboot after every settings change.

closed 2023-10-04 by D-an-W

Mario-vminion, 2023-11-26:
hi d-an-w .. i'm really new to the cam v3 and have no skill for programming..but are you willing to share you script file... i have also the pi zero 2 w and the cam v3 . i want to use this for the bird watch..
thanks

D-an-W, 2023-11-26:
> hi d-an-w .. i'm really new to the cam v3 and have no skill for programming..but are you willing to share you script file... i have also the pi zero 2 w and the cam v3 . i want to use this for the bird watch.. thanks

I’m away from home at the moment but will try my best to help when I get home, I basically followed the steps here though.