motioneye-project/motioneye #2833: S3 upload fails if there is a space in the file name

Clumpton, 2023-09-30:
Version 0.42.1
Debian Bullseye on RPi4
S3 upload configured and testing OK (to IDrive e2)
File naming set to %Y-%m-%d/%Y-%m-%d %H-%M-%S
Upload fails with:
`No such file or directory: '/var/lib/motioneye/Camera1/2023-09-30/2023-09-30'` 
File name is truncated at the space between date and time.
Current workaround, use _ instead of the space

MuhammadAbrar1984, 2023-10-01:
Not sure what exactly is being used underlying but spaces are normally required to be escapped. In the command line, paths with space can be used by enclosing them in double quotes. On the web, spaces are escaped by replacing space with "%20"

Clumpton, 2023-10-01:
> Not sure what exactly is being used underlying but spaces are normally required to be escapped. In the command line, paths with space can be used by enclosing them in double quotes. On the web, spaces are escaped by replacing space with "%20"

Thanks for that. However the software should either deal with spaces for S3 upload (as it does for local file naming) or issue a warning that spaces are not allowed in S3 uploads.

zagrim, 2023-10-16:
Most things should work even with spaces, but I'd suggest trying out the current dev branch version (Python3 support, plenty of fixes), since spaces and webhooks seem to be a common issue in the older releases.
Here are the instructions for installing the dev branch version: https://github.com/motioneye-project/motioneye/tree/dev#installation

MichaIng, 2026-02-08:
I'll mark this as closed. Feel free to reopen if the issue persists with current motionEye 0.43.y.