motioneye-project/motioneye #2834: Take picture with proximity sensor

3DFarmer, 2023-10-01:
Hi, I'm looking to use motioneye to create a dataset for ai training. I want to install a inductive sensor and to a pi's GPIO pin and have motioneye take one picture every time the sensor is activated. This sensor will be hooked up to a 3D printer, but the machine is close source so I can't use octoprint or other 3D printer servers. I love motioneye interface and ease of use.

Could this be accomplished?

git-tail-d, 2024-03-03:
Hello,

@3DFarmer 

i found a resolution.

you can take pictures with this command:

curl -sf http://192.168.X.Y:7999/1/action/snapshot

BR

JamBalaya56562, 2026-11-07:
Closing as answered, as the command above takes a picture. Checked with 0.45.0a1 and motion 4.7.1 on a Raspberry Pi, with two things to know:

- Still Images must be on with the capture mode set to Manual. With no image file name set, the command replies `Done` but saves nothing.
- By default this port only answers on the same device, so run the sensor script there: `curl http://127.0.0.1:7999/1/action/snapshot`

closed 2026-11-07 by JamBalaya56562