motioneye-project/motioneye #2855: Videos Sometimes Have Blocky or Blurred Areas

DC651F82053AD314, 2023-10-11:
Hi,

I'll explain my setup, then I'll list the intermittent issue I'm seeing...

I recently upgraded my security camera and Ubuntu-based computer running MotionEye. I'm running an SV3C POE PTZ 4k IP camera and connecting to it in MotionEye via the RTSP stream. The computer is a [Protectli VP2410](https://protectli.com/product/vp2410/), which has plenty of power for capturing the 4k stream, and all gigabit connections with an Intel Celeron quad core 2 GHz (Burst up to 2.7 GHz).

**Here is my MotionEye setting/setup**:
motionEye Version: 0.42.1
Motion Version: 4.3.2
OS Version: Ubuntu 22.04
Video Resolution: (Custom) 2560 X 1920
RTSP Connection Address: rtsp://192.168.1.22:554/11 
Video Device Frame Rate: 30 FPS
Video Streaming Frame Rate: 30 FPS
Video Streaming Quality: 90%
Movie Passthrough: Enabled
Movie Quality: 100%
Recording Mode: Motion Triggered

**Here's the IP camera output settings**:
Video Format: 60Hz
Video Coding: H265 main profile
Resolution: 2560 X 1920
Bit Rate: 2048
Maximum Frame Rate: 30 FPS
Key frame interval: 100
Bit rate control: VBR

First, I want to mention that at first the RTSP stream would not work correctly at all due to a known libmicrohttpd package 'chunker' issue -- I have downgraded it to libmicrohttpd12=0.9.72-2 in order to get it working correctly, and now, for the most part, it works great.  I found the answer to fix that here https://github.com/motioneye-project/motioneye/issues/2267 

However, I am seeing intermittent blurring/blocky output in some sections of the video in MotionEye.  The video is coming out clean from the camera, but when it gets to MotionEye, it seems to have trouble with (I think) compression in areas where there is a lot of detail in the video feed, like for example a bunch of leaves on the ground.

Please see these screenshots to understand the issue I'm describing in the video recordings from MotionEye:
[image]
[image]

Here is what it looks like when the issue isn't present:
[image]

As I mentioned, the issue only happens from time to time, not all of the time, and it seems like it's only happening in areas where there is a lot of tiny details to pick up.  I have verified that it's not the actual IP camera stream, it's only showing up in MotionEye.

If anyone has some advice for me to avoid seeing this issue, please let me know your suggestions.

Lastly, I would like to thank the MotionEye team for an awesome opensource offering! And thanks in advance for any help anyone can provide!

acyanic, 2023-11-08:
Hi, I did a web search and found this on [Openeye](https://answers.openeye.net/Troubleshooting/Frequently_Asked_Questions/Cameras/How_to_Fix_Camera_Smearing)

Quite a few things to try, but I would lower FPS to 15.

I have seen on IPcamtalk that 15 FPS is sufficient for surveillance monitoring.

Do you need Video streaming, mine's disabled ?

JamBalaya56562, 2026-11-12:
With Movie Passthrough, Motion writes the camera's H.265 packets as they arrive, so blocky areas in the movie are usually packets that were lost or late on the way, rather than something motionEye re-encodes. A lower bit rate or frame rate on the camera, as acyanic suggested, or its sub-stream, usually helps. As this was on motionEye 0.42.1 and Motion 4.3.2 with no follow-up since 2023, closing; please open a new issue if it still happens with current versions.

closed 2026-11-12 by JamBalaya56562