motioneye-project/motioneye #2858: Multiple Camera view

madakuoni, 2023-10-16:
Hallo,
I am using Motion Eye with 4 Rtsp Cameras. I am using Fullpage OS to view the live streams in Kiosk Mode. Is there a Chance to View them all 4 in Fullscreen Mode without the top bar? I can use fullscreen for each of them but i cant find a way for all 4 at once. Maybe there is a way to stream them at once with the Video stream Option ?

zagrim, 2023-10-16:
There is no "full screen with multiple cameras" mode. You'd have to build your own setup somehow, possibly using streaming URLs of the cameras.

madakuoni, 2023-10-16:
For the Next One i used this now and it works pretty well
Running on an apache2 Service you can View all 4 on an http surface for fullpage os.
```

<!DOCTYPE html>
<html lang="en">
<head>
  <meta charset="UTF-8">
  <meta name="viewport" content="width=device-width, initial-scale=1.0">
  <title>4 HTTP Streams im 2x2-Format</title>
  <style>
    .iframe-container {
      width: 50%; /* Jedes <iframe> nimmt 50% der Breite */
      float: left; /* Die <iframe>-Elemente nebeneinander positionieren */
    }
  </style>
</head>
<body>
  <div class="iframe-container">
    <iframe src="http:// xxx:8765/picture/1/frame" width="100%" height="100%" frameborder="0"></iframe>
  </div>
  <div class="iframe-container">
    <iframe src="http://xxx:8765/picture/2/frame" width="100%" height="100%" frameborder="0"></iframe>
  </div>
  <div class="iframe-container">
    <iframe src="http://xxx/picture/3/frame" width="100%" height="100%" frameborder="0"></iframe>
  </div>
  <div class="iframe-container">
    <iframe src="http://xxx/picture/4/frame" width="100%" height="100%" frameborder="0"></iframe>
  </div>
</body>
</html>

```

JamBalaya56562, 2026-11-07:
Closing as answered: as zagrim wrote above, there is no full screen mode for several cameras, and the page you built from the streams works for you. Thanks for sharing it.

closed 2026-11-07 by JamBalaya56562