motioneye-project/motioneye #2861: 502 Bad Gateway from home assistant plugin

michaelsage, 2023-10-20:
I'm not sure where the issue lies, I think it's probably within motioneye, but I don't have anything to really back this up.

I am running the latest dev version of motioneye on a pi 4 (bookworm). I have two cameras, they both work fine in the motioneye interface, however, when I add them to home assistant only one of them streams correctly. The other when opening the link on a new tab gives a 502: Bad Gateway. Any ideas on where I can look. I've tried turning debug logging on and that doesn't seem to report anything relevant. 

Thanks!

stormsnsnow, 2026-10-21:
I think this is due to SSL being invalid. Try disabling SSL in configuration settings.

MichaIng, 2026-10-21:
Currently there is another issue. This commit needs to be merged into a HA release tag, and then HA updated: https://github.com/home-assistant/core/commit/11de837

Next HA release 2026.10.3 or 2026.11.0 should have it.

stormsnsnow, 2026-10-21:
Oh ok thank you @MichaIng

JamBalaya56562, 2026-11-07:
The fix MichaIng mentioned above (home-assistant/core#181994) is now released in Home Assistant 2026.11.0. Could you check whether both cameras stream in Home Assistant with that version and motionEye 0.44? If one still gives 502, the Home Assistant log from that moment would help.