motioneye-project/motioneye #2863: Can't access web interface

yaomtc, 2023-10-21:
I had motioneyeOS installed initially, before I found out that it doesn't support Raspberry Pi Camera Module 3. MotionEye installed on a system with a newer kernel [should work](https://github.com/motioneye-project/motioneyeos/issues/2981). I figured motioneye_init would have started the web server, but unlike before, putting my Raspberry Pi's IP address in doesn't bring up the motionEye interface yet.

I looked at the journal but couldn't find anything specifically about the server having an issue. I can connect to it over SSH via that IP address (10.0.0.3), but when I try to access 10.0.0.3:8765 via my browser, it's unable to connect. http://10.0.0.3/ is an instant failure but with http://10.0.0.3:8765/ Firefox keeps trying for a while longer until it times out.

It's Raspbian Bullseye. The systemd service wasn't running at first (I thought the init script would have done that) so I enabling/started it, and here's the status:
```
● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2023-10-20 17:00:01 CDT; 1min 1s ago
   Main PID: 22187 (meyectl)
      Tasks: 5 (limit: 1595)
        CPU: 5.251s
     CGroup: /system.slice/motioneye.service
             ├─22187 /usr/bin/python3 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
             └─22233 /usr/bin/python3 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf

Oct 20 17:00:01 rpi systemd[1]: Started motionEye Server.
Oct 20 17:00:05 rpi meyectl[22187]: configure_logging cmd motioneye: False
Oct 20 17:00:05 rpi meyectl[22187]: configure logging to file: None
Oct 20 17:00:05 rpi meyectl[22187]:     INFO: hello! this is motionEye server 0.43.0
Oct 20 17:00:08 rpi meyectl[22187]:     INFO: cleanup started
Oct 20 17:00:08 rpi meyectl[22187]:     INFO: wsswitch started
Oct 20 17:00:08 rpi meyectl[22187]:     INFO: tasks started
Oct 20 17:00:08 rpi meyectl[22187]:     INFO: mjpg customer garbage collector has started
Oct 20 17:00:08 rpi meyectl[22187]:     INFO: server started
```

Still can't load. Also, this test:
```
$ ss -tulpn | grep 8765
tcp LISTEN 0 128 0.0.0.0:8765 0.0.0.0:*
```

zagrim, 2023-10-31:
If you try the following on the Pi, what do you get?
```
curl -I localhost:8765
curl -I 10.0.0.3:8765
sudo iptables -n -L INPUT |grep -E "(policy|8765)"
```

yaomtc, 2023-10-31:
```
$ curl -I localhost:8765
HTTP/1.1 200 OK
Server: motionEye/0.43.0
Content-Type: text/html; charset=UTF-8
Date: Tue, 31 Oct 2023 14:44:42 GMT
Etag: "da39a3ee5e6b4b0d3255bfef95601890afd80709"
Content-Length: 0
```

```
$ curl -I 10.0.0.3:8765
HTTP/1.1 200 OK
Server: motionEye/0.43.0
Content-Type: text/html; charset=UTF-8
Date: Tue, 31 Oct 2023 14:44:45 GMT
Etag: "da39a3ee5e6b4b0d3255bfef95601890afd80709"
Content-Length: 0
```

```
$ sudo iptables -n -L INPUT |grep -E "(policy|8765)"
Chain INPUT (policy ACCEPT)
```

yaomtc, 2023-11-01:
Ah. I can't use the camera. Guess that's #2799. Not sure how they got it working seemingly easily in https://github.com/motioneye-project/motioneyeos/issues/2981

zagrim, 2023-11-01:
There might even be more issues related to that camera module, but yes, I think #2799 is very much related. In the MEOS issue I think support for camera module 2 was confirmed, but based on the [comments](https://github.com/motioneye-project/motioneyeos/issues/2981#issuecomment-1463374481) I don't know if it really did work for v3. Anyway, [my comment](https://github.com/motioneye-project/motioneye/issues/2799#issuecomment-1689753179) on #2799 links to a PR which is supposed to have a fix for the issue of how to get the newer camera stack supported nicely, and if you are willing to test that I'm willing to try my best to guide you through it all even though I'm not really familiar with how that camera stuff works.