motioneye-project/motioneye #2865: Install failed: FileNotFoundError: [Errno 2] No such file or directory: 'curl-config'

FoobarTheFatPenguin, 2023-10-26:
I'm trying to install MotionEye on a Raspberry Pi400 running PiOS (Debian Bookworm). According to the instructions, I should do (as root):
```
python3 -m pip install 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'
```

That in and of itself gives an error message:
```
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

I had to create a venv and then use the command there for the process to actually start. But then I got the following error message:

```
Collecting pycurl
  Using cached pycurl-7.45.2.tar.gz (234 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [26 lines of output]
      Traceback (most recent call last):
        File "/tmp/pip-install-053h1_ax/pycurl_5e38433d77554466b35e1ca84f380b7e/setup.py", line 229, in configure_unix
          p = subprocess.Popen((self.curl_config(), '--version'),
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/lib/python3.11/subprocess.py", line 1024, in __init__
          self._execute_child(args, executable, preexec_fn, close_fds,
        File "/usr/lib/python3.11/subprocess.py", line 1901, in _execute_child
          raise child_exception_type(errno_num, err_msg, err_filename)
      FileNotFoundError: [Errno 2] No such file or directory: 'curl-config'
      
      During handling of the above exception, another exception occurred:
      
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-053h1_ax/pycurl_5e38433d77554466b35e1ca84f380b7e/setup.py", line 970, in <module>
          ext = get_extension(sys.argv, split_extension_source=split_extension_source)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-install-053h1_ax/pycurl_5e38433d77554466b35e1ca84f380b7e/setup.py", line 634, in get_extension
          ext_config = ExtensionConfiguration(argv)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-install-053h1_ax/pycurl_5e38433d77554466b35e1ca84f380b7e/setup.py", line 93, in __init__
          self.configure()
        File "/tmp/pip-install-053h1_ax/pycurl_5e38433d77554466b35e1ca84f380b7e/setup.py", line 234, in configure_unix
          raise ConfigurationError(msg)
      ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory: 'curl-config'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

I'm not exactly sure how to proceed from here.

zagrim, 2023-10-26:
`curl-config` should come with package `libcurl4-openssl-dev` which seems to be missing from the first step for ARM architecture from some reason. So please run `sudo apt --no-install-recommends install libcurl4-openssl-dev` and then try again the `pip install` command.
I wonder if Raspbian etc have that by default, or if it's because pycurl needed to be compiled on your system instead of having a precompiled package.

FoobarTheFatPenguin, 2023-10-26:
Thank you! That did the trick.

closed 2023-10-26 by FoobarTheFatPenguin