motioneye-project/motioneye #2866: No local Camera detection

cavernerg, 2023-10-26:
I installed dev via https://github.com/motioneye-project/motioneye/tree/dev on debian 11 which made me pretty happy because rtmp-stream from my reolink is finally working. But I lost the support for my local cameras. 

Detected cams:
`~# v4l2-ctl --list-devices
HD Webcam C615 (usb-0000:00:14.0-3.1.1):
        /dev/video2
        /dev/video3
        /dev/video8

WEB CAM: WEB CAM (usb-0000:00:14.0-4.3):
        /dev/video1
        /dev/video4
        /dev/video7`

Motioneye GUI: 
[image]

where did I go wrong?

cavernerg, 2023-10-27:
Cameras can be added by copying the old camera-n.conf from the older version. So it's definitely not in the hardware ;)

renanrider, 2023-11-17:
Try to this:

1 -  Install https://snapcraft.io/install/usb-reset/debian
2 - find devices ```lsusb```
3 - reset devices ```sudo usbreset 001/002```
4 - Add permission ``` sudo chmod 777 /dev/video0```
5 - Check if solved

csolisr, 2025-05-29:
I'm currently having a very similar problem - motionEye version 0.43.1, motion version 4.5.1 - and V4L2 remains undetected, even if I have granted the `motioneye` user access to the `video` group, and `motion` in standalone mode can access the camera with no problems. Logs below (translated from Spanish):

```
INFO: hello! this is a server for motionEye 0.43.1b1
DEBUG: using motion version 4.5.1
DEBUG: using ffmpeg version 5.1.6-0+deb12u1
DEBUG: listing config dir /var/www/motioneye...
DEBUG: Using selector: EpollSelector
INFO: cleanup started
INFO: wsswitch has started
INFO: tasks started
INFO: The mjpg garbage collector has started
INFO: server started
DEBUG: listing cameras
DEBUG: listing V4L2 devices
DEBUG: v4l2-ctl error: b''
```

SolAZDev, 2025-11-23:
Can confirm it's happening in 0.43.1b4, I just saw the new b5 release, but similarly, motion and motioneye were added to the video group, but the webui doesn't detect them

https://github.com/YunoHost-Apps/motioneye_ynh/issues/26#issuecomment-3567588644

JamBalaya56562, 2026-11-13:
For the YunoHost package, this was fixed in YunoHost-Apps/motioneye_ynh#28: its service hid the video devices (`PrivateDevices=yes`). In general, `v4l2-ctl error: b''` in the log means the `v4l2-ctl` call itself failed (it also shows when `v4l2-ctl` is not installed: package `v4l-utils`), and the camera list then stays empty. If it still happens, please post the output of `sudo -u motion v4l2-ctl --list-devices; echo $?` on the motionEye host (with the user motionEye runs as).