motioneye-project/motioneye #2867: DEV - Port 80 not working

cavernerg, 2023-10-27:
When I set "PORT = 80" in the motioneye.conf, the GUI is not reachable (after restart). Port 8765 is working fine.

cavernerg, 2023-10-31:
The port isn't even in use, when I check it via netstat. (I just substitute the number so lower or upper cases aren't the thing ;-))

zagrim, 2023-11-01:
Would you please provide the logs I mentioned above, I can't help if I'm not seeing what's happening there?

frank-que, 2023-11-16:
Hello @zagrim, I have the same problem, with port 8765 everything works correctly, but not with port 80.
I think it's a permissions problem... here is the log:

Nov 16 11:49:22 rpi5-motioneye systemd[1]: motioneye.service: Consumed 3.684s CPU time.
Nov 16 11:49:22 rpi5-motioneye systemd[1]: motioneye.service: Failed with result 'exit-code'.
Nov 16 11:49:21 rpi5-motioneye systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Nov 16 11:49:20 rpi5-motioneye meyectl[926]: AttributeError: 'NoneType' object has no attribute 'dumps'
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:   File "/usr/lib/python3.11/multiprocessing/queues.py", line 371, in put
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:   File "/usr/lib/python3.11/multiprocessing/pool.py", line 271, in __del__
Nov 16 11:49:20 rpi5-motioneye meyectl[926]: Traceback (most recent call last):
Nov 16 11:49:20 rpi5-motioneye meyectl[926]: Exception ignored in: <function Pool.__del__ at 0xffffb084db20>
Nov 16 11:49:20 rpi5-motioneye meyectl[926]: PermissionError: [Errno 13] Permission denied
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:     sock.bind(sockaddr)
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:   File "/usr/local/lib/python3.11/dist-packages/tornado/netutil.py", line 162, in bind_sockets
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:               ^^^^^^^^^^^^^
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:     sockets = bind_sockets(
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:   File "/usr/local/lib/python3.11/dist-packages/tornado/tcpserver.py", line 183, in listen
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:     server.listen(
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:   File "/usr/local/lib/python3.11/dist-packages/tornado/web.py", line 2207, in listen
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:     application.listen(settings.PORT, settings.LISTEN)
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 463, in run
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:     run()
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 504, in main
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:     server.main(arg_parser, sys.argv[2:], command[:-6])
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/meyectl.py", line 337, in main
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:              ^^^^^^
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:     sys.exit(main())
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:   File "/usr/local/bin/meyectl", line 8, in <module>
Nov 16 11:49:20 rpi5-motioneye meyectl[926]: Traceback (most recent call last):
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:     INFO: mjpg customer garbage collector has started
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:     INFO: tasks started
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:     INFO: wsswitch started
Nov 16 11:49:20 rpi5-motioneye meyectl[926]:     INFO: cleanup started

frank-que, 2023-11-16:
The problem come because port under 1024 are privileged (you need root permission to use)

I've bypassed the problem in this way:

install nginx: "sudo apt-get install nginx"

Remove or comment out any existing server section from file "/etc/nginx/sites-available/default" and add this:

server {
    listen 80;
    server_name _;

    location / {
        proxy_pass http://127.0.0.1:8765/;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header X-Forwarded-Proto $scheme;
        proxy_set_header X-Forwarded-Host $host;
        proxy_set_header X-Forwarded-Prefix /;
    }
}

I hope this can help someone.

zagrim, 2023-11-16:
@frank-que yes, something like that was one of the things I was suspecting.

JamBalaya56562, 2026-11-07:
Closing as answered. Checked with 0.45.0a1 on a Raspberry Pi: `port 80` makes the service fail with `PermissionError`, as it runs as the `motion` user and ports below 1024 need extra rights. Keep the default port and put a reverse proxy in front of it, as shown above with nginx.

closed 2026-11-07 by JamBalaya56562