motioneye-project/motioneye #2868: Unable to connect to IP camera Internec i6.4-C series

kfam-marcink, 2023-10-30:
Hello,
I'm trying to connect to the rtsp stream from camera as in the topic but in the field camera nothing apperars. I'm connecting from HAOS add-on. 
[image]
As I try to connect with the same link with VLC I have no problem getting to the video stream.
What am I doing wrong?
The link I'm using: rtsp://user:password@1xx.1xx.10.1xx/1/2/

acyanic, 2023-11-08:
just passing by, so my guess..
you have username and password in the URL as well as the boxes.
Try one (probably just in the boxes), not both.

geolr, 2024-01-24:
I think the url needs review, due to the @ it's not validated (red icon to the right in the textfield).

Try without `user:password@` in the url like:
rtsp://192.168.n.n/...

MichaIng, 2024-01-24:
Yes, URLs with embedded credentials are not even supported by modern browsers when present in the HTML document. In this case, when it is processed in the backend, this might not be the blocker, but we have the input fields to send them via HTTP authentication, which should be used (only) instead.

JamBalaya56562, 2026-11-12:
To sum up the answers: leave `user:password@` out of the URL, so it reads `rtsp://<ip>/1/2/`, and put the credentials in the Username and Password fields; motionEye passes them to Motion from there. As there was no follow-up since then, closing as answered.

closed 2026-11-12 by JamBalaya56562