motioneye-project/motioneye #2869: motioneye.service: Failed

filippecha, 2023-11-04:
Hello,
please advice for a beginner.

Raspberry zero W
+ this manual for getting Motioneye up and running
https://raspberrytips.com/install-motioneye-on-raspberry-pi/

after command "sudo service motioneye status" i get this error.
Thanks for any advice advice.

`motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor pre>
     Active: failed (Result: exit-code) since Sat 2023-11-04 15:52:27 CET; 1min>
    Process: 360 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye>
   Main PID: 360 (code=exited, status=255/EXCEPTION)
        CPU: 14.135s

lis 04 15:51:03 raspberrypi systemd[1]: Started motionEye Server.
lis 04 15:52:23 raspberrypi meyectl[360]: configure_logging cmd motioneye: False
lis 04 15:52:23 raspberrypi meyectl[360]: configure logging to file: None
lis 04 15:52:24 raspberrypi meyectl[360]:     INFO: hello! this is motionEye se>
lis 04 15:52:24 raspberrypi meyectl[360]: CRITICAL: log directory "/var/log" do>
lis 04 15:52:27 raspberrypi systemd[1]: motioneye.service: Main process exited,>
lis 04 15:52:27 raspberrypi systemd[1]: motioneye.service: Failed with result '>
lis 04 15:52:27 raspberrypi systemd[1]: motioneye.service: Consumed 14.135s CPU>
lines 1-15/15 (END)...skipping...
* motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2023-11-04 15:52:27 CET; 1min 34s ago
    Process: 360 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=255/EXCEP>
   Main PID: 360 (code=exited, status=255/EXCEPTION)
        CPU: 14.135s

lis 04 15:51:03 raspberrypi systemd[1]: Started motionEye Server.
lis 04 15:52:23 raspberrypi meyectl[360]: configure_logging cmd motioneye: False
lis 04 15:52:23 raspberrypi meyectl[360]: configure logging to file: None
lis 04 15:52:24 raspberrypi meyectl[360]:     INFO: hello! this is motionEye server 0.43.0
lis 04 15:52:24 raspberrypi meyectl[360]: CRITICAL: log directory "/var/log" does not exist or is not writable
lis 04 15:52:27 raspberrypi systemd[1]: motioneye.service: Main process exited, code=exited, status=255/EXCEPTION
lis 04 15:52:27 raspberrypi systemd[1]: motioneye.service: Failed with result 'exit-code'.
lis 04 15:52:27 raspberrypi systemd[1]: motioneye.service: Consumed 14.135s CPU time.
~
`

aurelien31430, 2023-11-06:
Bonjour, je suis confronté au meme problème depuis 2 jours...

zagrim, 2023-11-07:
This: `CRITICAL: log directory "/var/log" does not exist or is not writable`
How are the permissions set for `/var/log`? Please try with `ls -dl /var/log` and post the output.

Also, what value, if any, do you have set for `log_path` on `/etc/motioneye/motioneye.conf`?

patrick-GH, 2023-11-09:
Hello, I'm having the same issue as well after switching to dev branch
log_path: '/var/log`

On a new install with dev branch it's now: `/var/log/motion`
'drwxr-xr-x  8 root root  4.0K Nov  8 14:56 log'

I'm getting a different error now though

meyectl[331]:     sock.bind(sockaddr)
meyectl[331]: PermissionError: [Errno 13] Permission denied
meyectl[331]: Exception ignored in: <function Pool.__del__ at 0xb32ecc40>
meyectl[331]: Traceback (most recent call last):
meyectl[331]:   File "/usr/lib/python3.9/multiprocessing/pool.py", line 268, in __del__
meyectl[331]:   File "/usr/lib/python3.9/multiprocessing/queues.py", line 372, in put
meyectl[331]: AttributeError: 'NoneType' object has no attribute 'dumps'
systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE

zagrim, 2023-11-09:
Well, MotionEye needs to be allowed to write to the logging directory whatever it is. Earlier ME ran as root, but the systemd service installed by default by `motioneye_init` on systems that have systemd runs ME as user "motion", so for in an upgrade over existing setup one needs to 
1) Change the `log_path` on `/etc/motioneye/motioneye.conf` to not be `/var/log` (since that needs to be owned by root user), the default for new installs is `/var/log/motioneye` so that's probably a good choice. If it already was something like `/var/log/motion` there's probably no need to change it.
2) If you changed the `log_path`, create the new log directory with `sudo mkdir /var/log/motioneye` (substitute your choice of directory if needed).
3) Assign the ownership of logging directory to "motion" user with `sudo chown motion /var/log/motioneye` (substitute your choice of directory if needed).

For a new install I would assume permissions be ok (please inform me if it is not the case), but if they are not, the command in the third step above should fix it.

JamBalaya56562, 2026-10-31:
With the next release, an old `log_path /var/log` no longer stops the server: logging to files is off unless `log_to_file true` is set (#3394), and the log directory is only checked when it is on (#3465). Logs then go to the journal (`journalctl -u motioneye`).

Checked with an old config (`log_path /var/log`, no `log_to_file`) as a non-root user: 0.44.0 stops with `log directory "/var/log" does not exist or is not writable`, the current dev starts and answers. Closing as completed.

closed 2026-10-31 by JamBalaya56562