motioneye-project/motioneye #2871: Colour shaded blocks/artefacts on the preview window

xj25vm, 2023-11-09:
I have a problem with a fresh install of MotionEye. The preview window has blocks/artefacts of colour shades - sometimes green, sometimes red - about 1/2 to 1/3 of the windows size - moving across the screen all over the places. Almost like the video is corrupted. The specs of my setup are:

OS: Slackware 15.0
Kernel: 6.1.29
FFmpeg 5.1.3
Motion 4.5.0
Motioneye 0.42.1
Camera: Hikvision Hilook IPC-T250H-MU - 5MP / PoE / H264

Things I have tried:

1. Plugged the camera into another Motioneye server which has been running fine for a year - it works perfectly fine, with no artefacts. The spec of this server is: 
    Slackware 15.0
   Motion 4.3.1
   Motioneye 0.42.1
   FFmpeg 4.4.1

2. I have tried to replicate on the new/bad server the same software versions for all of the above. The preview window is still showing artefacts.

3. I have tried the good PoE switch from the good server - no difference

4. I have re-installed the bad server 3 times from scratch - once with Slackware 14.2, twice with Slackware 15.0

5. I have tried 3 different versions of Motioneye, 4 versions of Motion, 4 versions of FFmpeg. No difference.

6. I have connected directly to the Motion camera stream on the problematic server from a browser - the stream is perfect - there are not artefacts. This would suggest something is happening inside Motioneye somewhere

7. I tried Chrome and Firefox for loading Motioneye preview windows - no difference.

8. I re-installed everything again on a completely different hardware - a laptop - and made no difference.

I can't spot anything in the logs - there is the message "error capturing first image" in motion log - but I assume that is not relevant - as streaming directly from motion works fine:

motion.log

[0:motion] [NTC] [ALL] motion_startup: Motion 4.5.0 Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1:Camera1] [ERR] [ALL] motion_init: Error capturing first image

motioneye.log

2023-11-09 17:58:19: [motioneye]     INFO: hello! this is motionEye server 0.42.1
2023-11-09 17:58:22: [motioneye]     INFO: cleanup started
2023-11-09 17:58:22: [motioneye]     INFO: wsswitch started
2023-11-09 17:58:22: [motioneye]     INFO: tasks started
2023-11-09 17:58:22: [motioneye]     INFO: mjpg client garbage collector started
2023-11-09 17:58:22: [motioneye]     INFO: server started

I am really at a loss at what else I can try. Any suggestions are most welcome.

reemoteg, 2023-11-22:
xj25vm,
I experienced the same problem after upgrading from Ubuntu 20.04 to 22.04.  The problem has something to do with libmicrohttpd12.  The issue was fixed by downgrading libmicrohttpd12 from 0.9.75-3 to 0.9.72-2.

I found the answer here: https://github.com/motioneye-project/motioneye/issues/2401

Good luck.

xj25vm, 2023-11-30:
Thank you very much for that. After checking #2401, it does look exactly like the issue I am having. Unfortunately I had to give up and clone another Motioneye install I have running on a different server - as I ran out of time after 4 days of troubleshooting. I will try your suggestion the next time I install or upgrade a Motioneye server. Thank you again for responding