motioneye-project/motioneye #2873: thumbs are generated but not displayed in the preview

StefanWausH, 2023-11-13:
Hello everyone. I have Motioneye running on a RPI4 compute module together with the Dietpi distro and a network camera from Dahua connected. The thumbs of the individual videos are generated, but not displayed in the preview window. In earlier tests I had an original raspi-cam. Everything worked fine there.
Any ideas?
[image]
[image]

zagrim, 2023-11-13:
How have you installed MotionEye? This looks like an issue with recent Pillow versions that was just fixed in the dev branch (Python3 support) and also in edge docker images.
So basically the fix should be to upgrade using the dev version by either following [these instuctions](https://github.com/motioneye-project/motioneye/blob/dev/README.md) or [edge image from here](https://github.com/motioneye-project/motioneye/pkgs/container/motioneye).

StefanWausH, 2023-11-14:
The installation was carried out via the Dietpi distribution. https://dietpi.com/#home 
https://github.com/MichaIng is one of the developers. :) 
I tried it with Debian Bullseye as well as with the new Debian Bookworm version. The result is the same. And it already worked with it, which surprises me the most. I'll do the upgrade from your link again and report back. 
Thanks for your feedback for now

StefanWausH, 2023-11-14:
It worked...creepy.
I probably did the right thing at the time, i.e. carried out the update.
Thanks for the tip and have a nice day!
Greetings, Stefan

closed 2023-11-14 by StefanWausH