motioneye-project/motioneye #2875: Latest image for command to run on motion

manno-xx, 2023-11-18:
Hi!

Is there a way to get the latest image that was saved after a motion event?

I was hoping to send the last saved image to a web server somewhere without having to write some server0side logic. So, whatever the filename is on the RPi (running motioneyeOS currently) to, for instant, 'recent.jpg' on a webserver I host a site on.

I guess I can the command to work somehow (.sh probably) but I cannot find a reference to the most recent recorded image.

The File Storage -> Upload Media Files is not an option as it uploads everything and there is no telling what the name of the latest is.

Kind regards,
Manno

zagrim, 2023-11-19:
At least in my setup there's a symlink `lastsnap.jpg` in each camera directory pointing to the most recent image file.

manno-xx, 2023-11-19:
Thanks @zagrim! Indeed, Had not thought about looking there using ssh. Somehow I hoped to find the answer in the web interface, but this works!

JamBalaya56562, 2026-11-07:
Closing as answered, as you confirmed above: the `lastsnap.jpg` symlink in each camera directory points to the most recent image. Please reopen if you need something else here.

closed 2026-11-07 by JamBalaya56562