motioneye-project/motioneye #2876: [HOW-TO] stop motioneye from resizing network camera

Fisch64, 2023-11-20:
I got a picamera v3 setup as a network camera to make it work in motioneye. Its running in 3258x1832 and showing perfect quality when Im accessing it directly. Motioneye however butchers the resolution down to 640x480 and sets the image to 4:3. I cant find any way to change this behaviour in motion.conf, motioneye.conf or camera.conf. Any help would be appreciated.

ferociousdiablo, 2024-04-27:
You need to add height and width into the camera-1.conf file (change the 1 if not first camera)
e.g
height 1832
width 3258

mtewes, 2025-07-30:
Have the same issue. Adding height and width to camera-1.conf doesn't work, this file gets overwritten when changing something to the config from the web GUI.

zagrim, 2025-08-02:
@Fisch64 @mtewes are you not able to adjust the resolution to your liking in the UI? I've got the settings shown below for my network cameras so I can match them with the camera stream:

[image]

zagrim, 2025-08-02:
"Extra Options" is suitable only for those settings MotionEye UI isn't providing controls for, because all other settings will be reset by ME when settings are saved next time with the values that come either from the UI directly or that are derived by some logic from the settings in the UI.

Fisch64, 2025-08-02:
@zagrim It works now. Dont ask me how, it's been quite some time :D

closed 2025-08-02 by Fisch64