motioneye-project/motioneye #2891: Motioneye 0.43 No streaming popup on german

OberfragGER, 2023-12-13:
motionEye Version | 0.43.0
Motion Version | 4.6.0
OS Version | Linux 6.5.11-4-pve

I installed motioneye with this instructions. https://github.com/motioneye-project/motioneye/tree/dev#installation
When u change the language to german and click on

Snapshot URL
Streaming URL
Embed URL

No popup with the regarding popup appears.

TiboTake, 2023-12-23:
Hey, it´s a little Bug.

So set the lang to english and the popup is back.

MichaIng, 2023-12-23:
Can you show the service logs?
```sh
journalctl -u motioneye
```
or if no error is visible there, for errors in browser console?

I know it is an issue if a newly added language has either no frontend or backend translation files at all, but the German translation should be complete.

OberfragGER, 2023-12-25:
> Can you show the service logs?
> 
> ```shell
> journalctl -u motioneye
> ```
> 
> or if no error is visible there, for errors in browser console?
> 
> I know it is an issue if a newly added language has either no frontend or backend translation files at all, but the German translation should be complete.

Sorry. no error visible. Just no popup. I switched just one minite back from en to de and clicked on the strean-url/ snapshot url.
[image]

joournalctl shows just:

`Dec 25 21:55:39 motioneye motion[20451]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Dec 25 21:55:39 motioneye motion[20451]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Dec 25 21:55:39 motioneye motion[20451]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-3.conf
Dec 25 21:55:39 motioneye motion[20451]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Dec 25 21:55:39 motioneye motion[20451]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.6.0 Started
Dec 25 21:55:39 motioneye motion[20451]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Dec 25 21:55:39 motioneye motion[20451]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)`

OberfragGER, 2023-12-25:
> Hey, it´s a little Bug.
> 
> So set the lang to english and the popup is back.

I know. I did so but i did update the info for the bug.

MichaIng, 2023-12-27:
Thanks, the error trace from the browser console should do to find the issue. I'll have a look into it.

Marijn0, 2024-06-27:
I think I've found the problem.

First of all, this problem is not only with the German translation but with almost all translations except a few, including English.

The problem is in the function runAlertDialog in main.js:

function runAlertDialog(message, onOk, options) {
 var params = {
   title: message,
   buttons: i18n.gettext('bone'),
   onOk: onOk
 };

But why is the translation of 'bone' used to determine the name of the button to be called?
"(buttons: i18n.gettext('bone'),)"

The translation of 'bone' in the English translation is: 'ok'. This is also why the English translation works.
The intention is to call the following:
https://github.com/motioneye-project/motioneye/blob/72af3711d28175669ca06ab09913b788a5fd8617/motioneye/static/js/ui.js#L1034

But when you use another language problems start to arise 😫
I think: "buttons: i18n.gettext('bone'),"
should be changed to: "buttons: 'ok'"
As it also happens in runConfirmDialog:

(for comparison)

function runConfirmDialog(message, onYes, options) {
 var params = {
   title: message,
   buttons: 'yesno',
   onYes: onYes
 };

MichaIng, 2024-06-27:
Good find. We should change the identifier of the button to not depend on the localised text, but based on some ID. I'll have a look the next days, unless someone is faster.

MichaIng, 2024-06-30:
Fix merged: #3020

Many thanks @Marijn0! Indeed here the ID was translated, which of course was never meant to happen.

PR #3020 (2024-06-30): Make snapshot/streaming/embed (URLs) pop-ups work on all translations
Fixes #2891 

Changed the function runAlertDialog to not depend on the selected language.

@MichaIng 
I hope this is how you envisioned the solution. If I need to change anything, please let me know!
closed 2024-06-30 by MichaIng