motioneye-project/motioneye #2899: Embed URL asking user/pass - refuses URL setting of it - Beta

BKeyport, 2023-12-27:
Embed urls are asking for user/pass in the latest beta. 

use http://vidhost.local:8765/picture/1/frame/ - you get a popup asking for login. 
user http://user:pass@vidhost.local:8765/picture/1/frame/ - you get nothing. 

Old versions installed through MotioneyeOS never asked for a user/pass on this.

MichaIng, 2023-12-27:
Thanks for reporting. If I remember correctly, modern browsers, especially Chromium-based ones, deny accepting embedded credentials for camera streams or in general for resources requested by the HTML document. Probably modern libraries do the same for resources requested by/passed through the backend.

Did you add this camera as regular network camera or as direct P2P stream (without motion detection)?

BKeyport, 2023-12-27:
Its added as a network camera. 

For reference, I'm working with electron on Raspberry Pi, using Motioneye to convert RTSP stream to my MagicNirror install. I've had little success in converting the stream directly, and with the loss of OMX, I've had other things rigged in place to make a viewer from scratch.

BKeyport, 2023-12-28:
Oh, and for testing purposes, this is the snippet that ran that url: 
```<iframe src="http://vidhost.local:8765/picture/1/frame/?timestamp=1703805114" frameborder="0" class="embeded"></iframe>``` - MMM-EmbedURL is including the timestamp and creating the iframe.

MichaIng, 2023-12-29:
Ah right, it is about an URL to embed into another website. I thought it was about the camera view in motionEye GUI, and with "embed URL" you meant that user:pass are embedded.

In this case this pretty much makes sense that it does not work with embedded user:pass. When opening the website with this `iframe` with embedded credentials, you should see a related error in browser console, that access to such URLs is denied. This is a browser decision and outside of our control.

So I guess with motionEyeOS it also only works when using no surveillance user password. I remember and found this being reported already. Let me close this issue in favour of the older one, where it was discussed whether to not require any authentication until a surveillance user password was set, or whether to add another toggle to explicitly disable authentication for motionEye camera streams/snapshots: #2686
And indeed there it was reported as well that motionEyeOS did not require any authentication, at least not without surveillance user password.

closed 2023-12-29 by MichaIng