motioneye-project/motioneye #2901: Development environment / live reload

sevimuelli, 2023-12-27:
Hi
There is not much documentation concerning development of motionEye. What is the best way to run motionEye for development? Some way to have the project open in VS Code and continuously update the changes. Ideally, I could run it inside a container. I saw that inside server.py, when starting the Tornado webserver, there is an argument "debug", which is kinda doing this. But there is no way to set this under normal operation.
What is the normal approach the developer here use? Any help is appreciated.

MichaIng, 2023-12-27:
Python does not seem to have a native autoreload feature, which would be required here.

I tend to just restart motionEye when testing changes:
```sh
systemctl restart motioneye
```

zagrim, 2024-02-03:
I tend to use the following manual "reload" cycle (simplified version here, as if not using Python venv):
Rebuild: `rm -r build/ motioneye.egg-info/; python3 -m pip install .`
Start: `~/.local/bin/meyectl startserver -c path/to/my/development/config`
Terminate server and repeat after changes to code.

Omumarvaishya, 2024-10-07:
i have facing the same issue could you please mention some steps to establish dev environment?

JamBalaya56562, 2026-11-12:
MichaIng and zagrim described how changes are tested here: install the checkout with pip (into a venv) and restart `meyectl startserver` after each change; there is no live reload. Closing as answered.

closed 2026-11-12 by JamBalaya56562