motioneye-project/motioneye #2904: Docker install on Buster

Kobbe1, 2023-12-30:
Hi,
great project!
I thought I give the 0.43 release a try in parallel on my production system (still running buster).
I disabled the old 0.42 motioneye service temporarily, and then tried below command:

`docker run --name="motioneye" -p 8765:8765 --hostname="motioneye" -v /etc/localtime:/etc/localtime:ro -v /etc/motioneye:/etc/motioneye -v /var/lib/motioneye:/var/lib/motioneye --restart=unless-stopped --detach=true ghcr.io/motioneye-project/motioneye:edge`

I am getting several permission errors, so I assumed I am maybe encountering [this](https://github.com/motioneye-project/motioneye/issues/2869). Following the 3 steps at the end did not really fix it, but then again I guess I am not actually running as user "motion" (I ran above command under my standard username, I guess I could try to just run the docker as user "motion"?).
I am pretty sure this is a permission issue/related to my old install of ME, but I am hoping someone can suggest a reasonable path forward, the old 0.42 install works very well so I am not too keen to mess it up completely - never change a running system.

I guess the docker wiki also needs an update but I think this was said before in some other issue.

Best,

-T

`pi@raspberrypi:~ $ docker logs motioneye`

`configure_logging cmd motioneye: False
configure logging to file: None
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/motioneye/meyectl.py", line 337, in main
    server.main(arg_parser, sys.argv[2:], command[:-6])
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 504, in main
    run()
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 433, in run
    start_motion()
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 363, in start_motion
    io_loop = IOLoop.current()
              ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/tornado/ioloop.py", line 274, in current
    loop = asyncio.get_event_loop()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/events.py", line 674, in get_event_loop
    self.set_event_loop(self.new_event_loop())
                        ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/events.py", line 695, in new_event_loop
    return self._loop_factory()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/unix_events.py", line 64, in __init__
    super().__init__(selector)
  File "/usr/lib/python3.11/asyncio/selector_events.py", line 50, in __init__
    super().__init__()
  File "/usr/lib/python3.11/asyncio/base_events.py", line 400, in __init__
    self._clock_resolution = time.get_clock_info('monotonic').resolution
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 1] Operation not permitted
Exception ignored in: <function BaseEventLoop.__del__ at 0x764257a8>
Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/base_events.py", line 692, in __del__
    _warn(f"unclosed event loop {self!r}", ResourceWarning, source=self)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 422, in __repr__
    f'closed={self.is_closed()} debug={self.get_debug()}>'
                                       ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 1941, in get_debug
    return self._debug
           ^^^^^^^^^^^
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'
configure_logging cmd motioneye: False
configure logging to file: None
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/motioneye/meyectl.py", line 337, in main
    server.main(arg_parser, sys.argv[2:], command[:-6])
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 504, in main
    run()
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 433, in run
    start_motion()
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 363, in start_motion
    io_loop = IOLoop.current()
              ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/tornado/ioloop.py", line 274, in current
    loop = asyncio.get_event_loop()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/events.py", line 674, in get_event_loop
    self.set_event_loop(self.new_event_loop())
                        ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/events.py", line 695, in new_event_loop
    return self._loop_factory()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/unix_events.py", line 64, in __init__
    super().__init__(selector)
  File "/usr/lib/python3.11/asyncio/selector_events.py", line 50, in __init__
    super().__init__()
  File "/usr/lib/python3.11/asyncio/base_events.py", line 400, in __init__
    self._clock_resolution = time.get_clock_info('monotonic').resolution
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 1] Operation not permitted
Exception ignored in: <function BaseEventLoop.__del__ at 0x763e57a8>
Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/base_events.py", line 692, in __del__
    _warn(f"unclosed event loop {self!r}", ResourceWarning, source=self)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 422, in __repr__
    f'closed={self.is_closed()} debug={self.get_debug()}>'
                                       ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 1941, in get_debug
    return self._debug
           ^^^^^^^^^^^
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'
configure_logging cmd motioneye: False
configure logging to file: None
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/motioneye/meyectl.py", line 337, in main
    server.main(arg_parser, sys.argv[2:], command[:-6])
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 504, in main
    run()
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 433, in run
    start_motion()
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 363, in start_motion
    io_loop = IOLoop.current()
              ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/tornado/ioloop.py", line 274, in current
    loop = asyncio.get_event_loop()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/events.py", line 674, in get_event_loop
    self.set_event_loop(self.new_event_loop())
                        ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/events.py", line 695, in new_event_loop
    return self._loop_factory()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/unix_events.py", line 64, in __init__
    super().__init__(selector)
  File "/usr/lib/python3.11/asyncio/selector_events.py", line 50, in __init__
    super().__init__()
  File "/usr/lib/python3.11/asyncio/base_events.py", line 400, in __init__
    self._clock_resolution = time.get_clock_info('monotonic').resolution
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 1] Operation not permitted
Exception ignored in: <function BaseEventLoop.__del__ at 0x764c57a8>
Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/base_events.py", line 692, in __del__
    _warn(f"unclosed event loop {self!r}", ResourceWarning, source=self)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 422, in __repr__
    f'closed={self.is_closed()} debug={self.get_debug()}>'
                                       ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 1941, in get_debug
    return self._debug
           ^^^^^^^^^^^
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'`

MichaIng, 2023-12-30:
Thanks for reporting. The final actual error that is thrown in a loop is:
```
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'
```
Strange about it is that this attribute is requested solely within the `asyncio` module, not by us.

The this error looks more like an attempt to log/warn about another one happened before:
```
  File "/usr/lib/python3.11/asyncio/base_events.py", line 400, in __init__
    self._clock_resolution = time.get_clock_info('monotonic').resolution
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 1] Operation not permitted
``
As far as I understand the docs about this, it is not even the system clock, but about an internal clock/timer started internally, in this case likely to time the I/O loop iterations: https://docs.python.org/3/library/time.html?highlight=get_clock_info
So no idea how a permissions error can happen in this case. `asyncio` and `time` are both Python 3 internal modules, so nothing where 3rd party updates/dependencies should have any effect on.

The issue you linked was about a bare metal motionEye instance on RPi executed via systemd unit, not a Docker container, hence entirely different. You do not need to (or better must not) run the container itself as `motion` user, especially since that one should not exist on your host system anyway. The container currently requires root permissions, and internally starts motionEye with its internal `motion` user: https://github.com/motioneye-project/motioneye/blob/dev/docker/entrypoint.sh

The RAEDME indeed needs to be updated for the new version. Your `docker run` however looks fine. The dirs of the volumes on host end exist?
Ah, although can you try to ship the volume and port options in your `docker run`? These are defined in the Dockerfile already: https://github.com/motioneye-project/motioneye/blob/dev/docker/Dockerfile#L40-L46
The two volumes are created in the Docker container's directory. I am currently not sure whether the declarations in the Dockerfile just override those you set via CLI, or whether they indeed can conflict.

Kobbe1, 2023-12-30:
Thanks for the help.
Tried as below, no luck. Sorry for spamming this issue with all these logs but I guess better too many than too few.
Strangely there are a few lines more at the start now (INFO). I also saw that the old ME restarts even though I disabled and stopped it:

```
sudo systemctl disable motioneye
sudo systemctl stop motioneye
```
Service says disabled but it still restarts after a while ?! :)

```
pi@raspberrypi:~ $ sudo systemctl status motioneye
* motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; disabled; vendor preset: enabled)
   Active: active (running) since Sat 2023-12-30 23:26:29 CET; 39s ago
 Main PID: 10400 (meyectl)
    Tasks: 25 (limit: 1935)
...
```
I did the docker test right after I stopped old ME, and it only restarts after some minutes so probably that is not the issue, but still strange (or more likely my lack of knowledge and that behaviour is normal).

The folders do exist from the old ME instance, and I manually created /var/log/motioneye already before to no avail.

```
pi@raspberrypi:~ $ docker stop motioneye
motioneye
pi@raspberrypi:~ $ docker rm motioneye
motioneye
pi@raspberrypi:~ $ docker run --name="motioneye" --hostname="motioneye" -v /etc/localtime:/etc/localtime:ro --restart=unless-stopped --detach=true ghcr.io/motioneye-project/motioneye:edge56c7494cec7feae7776991e5048783215273d568876052ab8eef3d330fb771eb
pi@raspberrypi:~ $ docker logs motioneye
configure_logging cmd motioneye: False
configure logging to file: None
    INFO: hello! this is motionEye server 0.43.1b1
    INFO: motion not installed
    INFO: v4l-utils not installed
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/motioneye/meyectl.py", line 337, in main
    server.main(arg_parser, sys.argv[2:], command[:-6])
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 504, in main
    run()
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 433, in run
    start_motion()
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 363, in start_motion
    io_loop = IOLoop.current()
              ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/tornado/ioloop.py", line 274, in current
    loop = asyncio.get_event_loop()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/events.py", line 674, in get_event_loop
    self.set_event_loop(self.new_event_loop())
                        ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/events.py", line 695, in new_event_loop
    return self._loop_factory()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/unix_events.py", line 64, in __init__
    super().__init__(selector)
  File "/usr/lib/python3.11/asyncio/selector_events.py", line 50, in __init__
    super().__init__()
  File "/usr/lib/python3.11/asyncio/base_events.py", line 400, in __init__
    self._clock_resolution = time.get_clock_info('monotonic').resolution
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 1] Operation not permitted
Exception ignored in: <function BaseEventLoop.__del__ at 0x764d57a8>
Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/base_events.py", line 692, in __del__
    _warn(f"unclosed event loop {self!r}", ResourceWarning, source=self)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 422, in __repr__
    f'closed={self.is_closed()} debug={self.get_debug()}>'
                                       ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 1941, in get_debug
    return self._debug
           ^^^^^^^^^^^
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'
```

MichaIng, 2023-12-30:
Hmm, no systemd service should start automatically when you disable (and stop) it. Check
```sh
journalctl -u motioneye
```
to see when this happens, and try to find out what triggers it. For the container as well, which you explicitly set as `--restart=unless-stopped`, i.e. to never restart when it was explicitly stopped once.

However, I see you run the container as `pi` user. Please try to do this as root. I am not exactly sure which part exactly requires root permissions and why, but the "not permitted" error quite points into this direction.

Kobbe1, 2023-12-30:
No luck it seems. Maybe will fix it next year ;)
```
pi@raspberrypi:~ $ docker stop motioneye
motioneye
pi@raspberrypi:~ $ docker rm  motioneye
docker: 'srm' is not a docker command.
See 'docker --help'
pi@raspberrypi:~ $ docker rm motioneye
motioneye
pi@raspberrypi:~ $ sudo docker run --name="motioneye" --hostname="motioneye" -v /etc/localtime:/etc/localtime:ro --restart=unless-stopped --detach=true ghcr.io/motioneye-project/motioneye:edge
23ed4165d54d3d44da6aaff750879c7bade2b1735e3c0898de60b037fa6192fa
pi@raspberrypi:~ $ docker logs motioneye
configure_logging cmd motioneye: False
configure logging to file: None
    INFO: hello! this is motionEye server 0.43.1b1
    INFO: motion not installed
    INFO: v4l-utils not installed
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/motioneye/meyectl.py", line 337, in main
    server.main(arg_parser, sys.argv[2:], command[:-6])
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 504, in main
    run()
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 433, in run
    start_motion()
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 363, in start_motion
    io_loop = IOLoop.current()
              ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/tornado/ioloop.py", line 274, in current
    loop = asyncio.get_event_loop()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/events.py", line 674, in get_event_loop
    self.set_event_loop(self.new_event_loop())
                        ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/events.py", line 695, in new_event_loop
    return self._loop_factory()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/unix_events.py", line 64, in __init__
    super().__init__(selector)
  File "/usr/lib/python3.11/asyncio/selector_events.py", line 50, in __init__
    super().__init__()
  File "/usr/lib/python3.11/asyncio/base_events.py", line 400, in __init__
    self._clock_resolution = time.get_clock_info('monotonic').resolution
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 1] Operation not permitted
Exception ignored in: <function BaseEventLoop.__del__ at 0x764357a8>
Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/base_events.py", line 692, in __del__
    _warn(f"unclosed event loop {self!r}", ResourceWarning, source=self)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 422, in __repr__
    f'closed={self.is_closed()} debug={self.get_debug()}>'
                                       ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 1941, in get_debug
    return self._debug
           ^^^^^^^^^^^
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'
pi@raspberrypi:~ $
```

On the stuborn ME 0.42:
```
Dez 31 00:12:38 raspberrypi systemd[1]: Stopping motionEye Server...
Dez 31 00:12:43 raspberrypi systemd[1]: motioneye.service: Succeeded.
Dez 31 00:12:43 raspberrypi systemd[1]: Stopped motionEye Server.
Dez 31 00:21:28 raspberrypi systemd[1]: Started motionEye Server.
Dez 31 00:21:32 raspberrypi motion[21710]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - 
```
and

```
pi@raspberrypi:~ $ sudo systemctl status motioneye
* motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; disabled; vendor preset: enabled)
   Active: active (running) since Sun 2023-12-31 00:21:28 CET; 2min 3s ago
 Main PID: 21692 (meyectl)
    Tasks: 27 (limit: 1935)
   CGroup: /system.slice/motioneye.service
```

MichaIng, 2023-12-30:
```
sudo docker logs motioneye
```
😉
I am sure this is an entirely new image, container and volumes, compared to the one you did spin up as `pi` user.

Kobbe1, 2023-12-30:
Thanks for your patience, I actually thought about it and then did it wrong anyway :-) Although no difference either it seems? I did successfully docker rm before (without sudo), so not sure the sudo makes a difference, maybe the docker service always runs as root? But worth a try anyway.

```
pi@raspberrypi:~ $ systemctl cat motioneye
# /etc/systemd/system/motioneye.service
[Unit]
Description=motionEye Server

[Service]
ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
Restart=on-abort

[Install]
WantedBy=multi-user.target
pi@raspberrypi:~ $ systemctl status motioneye.timer
Unit motioneye.timer could not be found.
pi@raspberrypi:~ $ sudo systemctl status motioneye.timer
Unit motioneye.timer could not be found.
pi@raspberrypi:~ $ sudo systemctl cat motioneye
# /etc/systemd/system/motioneye.service
[Unit]
Description=motionEye Server

[Service]
ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
Restart=on-abort

[Install]
WantedBy=multi-user.target
```

```
Last login: Sun Dec 31 00:15:02 2023 from 192.168.178.xxx
pi@raspberrypi:~ $ sudo systemctl disable motioneye
pi@raspberrypi:~ $ sudo systemctl stop motioneye
pi@raspberrypi:~ $ docker rm motioneye
Error response from daemon: No such container: motioneye
pi@raspberrypi:~ $ sudo docker rm motioneye
Error response from daemon: No such container: motioneye
pi@raspberrypi:~ $ sudo docker run --name="motioneye" --hostname="motioneye" -v /etc/localtime:/etc/localtime:ro --restart=unless-stopped --detach=true ghcr.io/motioneye-project/motioneye:edge
080449d10eec39d32800469398e79a00baf8ca19b717f1ea576472a488f73f0c
pi@raspberrypi:~ $ sudo docker logs motioneye
configure_logging cmd motioneye: False
configure logging to file: None
    INFO: hello! this is motionEye server 0.43.1b1
    INFO: motion not installed
    INFO: v4l-utils not installed
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/motioneye/meyectl.py", line 337, in main
    server.main(arg_parser, sys.argv[2:], command[:-6])
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 504, in main
    run()
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 433, in run
    start_motion()
  File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 363, in start_motion
    io_loop = IOLoop.current()
              ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/tornado/ioloop.py", line 274, in current
    loop = asyncio.get_event_loop()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/events.py", line 674, in get_event_loop
    self.set_event_loop(self.new_event_loop())
                        ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/events.py", line 695, in new_event_loop
    return self._loop_factory()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/unix_events.py", line 64, in __init__
[... 184 lines cut ...]
  File "/usr/lib/python3.11/asyncio/events.py", line 695, in new_event_loop
    return self._loop_factory()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/unix_events.py", line 64, in __init__
    super().__init__(selector)
  File "/usr/lib/python3.11/asyncio/selector_events.py", line 50, in __init__
    super().__init__()
  File "/usr/lib/python3.11/asyncio/base_events.py", line 400, in __init__
    self._clock_resolution = time.get_clock_info('monotonic').resolution
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 1] Operation not permitted
Exception ignored in: <function BaseEventLoop.__del__ at 0x764557a8>
Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/base_events.py", line 692, in __del__
    _warn(f"unclosed event loop {self!r}", ResourceWarning, source=self)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 422, in __repr__
    f'closed={self.is_closed()} debug={self.get_debug()}>'
                                       ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 1941, in get_debug
    return self._debug
           ^^^^^^^^^^^
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'
pi@raspberrypi:~ $
```

MichaIng, 2023-12-30:
Okay you are probably right. Running non-root containers was not that easy yet IIRC and it seems the `pi` user just has the right permissions to control root containers.

I won't find the time to test it myself until next year either. So I wish you Happy New Year in advance, despite the motionEye issue, and I will check back and test it on a VM and RPi Zero W and/or RPi 2 next year.

JamBalaya56562, 2026-11-12:
Sorry that this was left open. Since 0.45.0b1 (#3484), the image runs as root unless another user is given with `docker run --user`, so host directories owned by root no longer give permission errors; [Running as a non-root user](https://github.com/motioneye-project/motioneye/blob/dev/docker/README.md#running-as-a-non-root-user) describes the `--user` way and the `chown` it needs. Closing this; please open a new issue if the current image still fails on that host.

closed 2026-11-12 by JamBalaya56562