motioneye-project/motioneye #2905: How to fix this mistake

Gurpreet83, 2023-12-31:
Hello.

Newbie here, apologies if this is a dumb thing to ask..please can you help with the below error.

gurp@raspberrypi:~ $ sudo python3 -m pip install --pre motioneye
Configuration file could not be loaded.
While reading from '/etc/pip.conf' [line  3]: option 'extra-index-url' in section 'global' already exists

sanch03, 2024-01-05:
You probably ran one of the commands more than once.
Run this: `cat /etc/pip.conf`
The output should look something like this:
```
[global]
extra-index-url=https://www.piwheels.org/simple/
break-system-packages=true
```

If you have duplicate extra-index-url entries, you need to edit the conf and remove them.

MichaIng, 2024-01-05:
I guess you ran the related command from the readme doubled. We should probably either add a `grep ... || sed ...` check for all of them, or just show the content of the config files.

keithy, 2024-09-25:
Since when has having duplicate extra-index-urls been a problem you used to be able to specify multiple entries on the command line

closed 2026-10-31 by Marijn0