motioneye-project/motioneye #2906: Install on Raspi 5

womihako, 2024-01-03:
Hello, 
first I will thanks all for making and support this very very nice project!!!  it run by me on raspi 4, but can't install on raspy 5. I tried installation instructions #2780 and other.
Can someone help me pls?
Thanks a lot

ddawley22, 2024-01-07:
I installed it yesterday on RPi 5 using the instructions posted here:

https://github.com/motioneye-project/motioneye/tree/dev

I might add that video recording doesn't make on this version ( 0.43.1b1) nor on another similar version from a year ago that I installed on a Libre le Potato.  The snapshots record but video doesn't.  It's not an issue for my situation but someone might want to check into it.

MichaIng, 2024-01-07:
> Use care in selecting the right directions because a misstep will break the process. I ended up starting over with a fresh install of bookworm because I messed up the process.

How did you break the process? It should be possible to redo/fix everything without doing a fresh install.

ddawley22, 2024-01-07:
I cross installed a part of the 32 bit instructions on my 64 bit version and I got errors trying to overwrite using the 64 bit instructions. I probably could have reversed the part I accidentally installed but since it was a fresh test to see if it would work, I decided a clean install wasn't a big deal.

On another note, regarding the video not working, I changed the Motion Detection Capture Before setting from 1 (default) to 2, and video recording started working.

MichaIng, 2024-01-07:
Do you remember the error you faced, and at which step exactly?

In case of the dependencies, the 32-bit instructions contain just less dependencies, so installing the ones for 64-bit afterwards should fix that part.

In case of adding the piwheels index, AFAIK it should be just ignored on 64-bit systems since the wheels it provides should be tagged a way that they are not pulled/installed from there, if for 32-bit only. Otherwise, you can just remove it:
```sh
sudo sed -i '/piwheels/d' /etc/pip.conf
```

ddawley22, 2024-01-07:
Thanks for that.  I don't remember exactly where I erred, sorry.

rahul1906, 2024-09-24:
@ddawley22  How Is the performance of motion eye on pi5 compared to pi4? Asking this since h264 h/w codec has been removed in pi5 and most cc cams use this codec.

JamBalaya56562, 2026-11-06:
motionEye installs and runs on a Raspberry Pi 5. Checked with 0.44.0 on Raspberry Pi OS trixie (Python 3.13, motion 4.7.1), following the [README on `main`](https://github.com/motioneye-project/motioneye/blob/main/README.md). One exception: on this image `get-pip.py` in step 2 fails with `Cannot uninstall pip`, as Debian's `python3-pip` is already installed, so skip that command and keep the `pip.conf` change. The server starts, and the Camera Module 3 gives a picture and records a movie through `libcamerify`, see [Troubleshooting](https://github.com/motioneye-project/motioneye/wiki/Troubleshooting#libcamera-camera-stack). Closing as completed; please reopen if the install still fails.

closed 2026-11-06 by JamBalaya56562