motioneye-project/motioneye #2907: Smearing and green video

McMagnus, 2024-01-03:
I first tried MotionEye on my QNap NAS running Debian 10.13, but (mostly during daytime) I saw very bad smearing on the lower part of the picture, so it triggered motion very often, like every other minute or so.

So I thought the NAS wasn't up to the job and tried running it in an oldish i7 laptop in virtualbox Lubuntu 20, and that went much better. So I made a new vbox (Lubuntu 22) in an old i3 PC that I could have online 24/7. But then the smearing started again. So i tried Lubuntu 20 in the i3 PC and now it works fine.

I'm sure the problem is related to motion, and not motioneye specifically, but I wanted to gather more info about this problem that I'd think make many people leave?

So, does anyone have more info? Is there some specific version of motion/motioneye or some other component that is the root cause? Or is there a combination?

McMagnus, 2024-01-03:
Found a possible solution here: https://github.com/motioneye-project/motioneye/issues/2871

Will try and verify.

JamBalaya56562, 2026-11-12:
Thanks for reporting back: #2871 fixed it on Lubuntu 22, and the dev branch on the QNAP NAS. Closing as resolved; if the smearing comes back with a current release, please open a new issue with the camera's stream settings and the Motion log.