motioneye-project/motioneye #2908: V4L2 device failed to open

Stormbringer906, 2024-01-03:
I have a RPI4 with a LovePI Camera Module that I'm trying to install MotionEye on.  I can run libcamera-still just fine.

When I try and setup a camera in MotionEye, I get a big list.  None of these work though.  I get V4L2 device failed to open.

[image]

Any ideas?

Thanks,
SB

kni-bo, 2024-01-04:
Hi,
I had the same problem. The solution is libcamerify see [here](https://github.com/motioneye-project/motioneye/issues/2812).

Greetings Michael

Stormbringer906, 2024-01-04:
Thank you VERY much.  I've been going in circles with this.

I have some old PI Zeros and an RPI 4 I've been playing with.  They now work GREAT!  On a fully updated OS!

SB

closed 2024-01-04 by Stormbringer906