motioneye-project/motioneye #2911: Fullscreen in user-mode not working correct.

TheRaven500, 2024-01-06:
Small issue:
If logged in as user "admin" all is working ok. But if logged in as "user" (view only), fullscreen is not working probably. It is not really fullscreen, looks more like window-mode or something like that.
Is this a config-problem or can someone else reproduce this?

JamBalaya56562, 2026-11-13:
I could not reproduce this on the current development version: in two browsers, the full screen button gives the same layout logged in as admin and as the surveillance user, apart from the configure button, and the full screen code has no admin-only branch. Closing; if it still happens on 0.45.0b1 or later, please reopen with the browser and a screenshot of both.

closed 2026-11-13 by JamBalaya56562

TheRaven500, 2026-11-13:
It was the same problem like here:
https://github.com/motioneye-project/motioneye/issues/3417 (closed)
It was a "user error". I just forgot to close this old issue sorry.