motioneye-project/motioneye #2914: Please suuport RTSPS Streams

FlexerJR, 2024-01-12:
Hi Devs. 

First of all thank you for MotionEye! It's a great tool I'm working with.

Unfortunately more and more Cameras will only be accessible with RTSPS with is a other protocol! Please make it possible so add Cameras with RTSPS Streams which is used by Bambulab X1C for example

Thank you and best regards

zagrim, 2024-01-13:
I was already about to suggest creating a feature request in Motion repository, but despite the docs lacking any mention on RTSPS, it seems that it should already be supported in the recent release(s) (see discussion on https://github.com/Motion-Project/motion/pull/1346 and the changes in https://github.com/Motion-Project/motion/pull/1368), starting with 4.4.0.

In ME, it is possible to put RTSPS URL for a network camera (after a trivial change in [handlers/config.py#L432](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/handlers/config.py#L432), but it seems that we do some more fine-grained protocol specific stuff in [utils/rtsp.py](https://github.com/motioneye-project/motioneye/blob/dev/motioneye/utils/rtsp.py) which would need to be adapted to the secure version of the protocol. Someone who has access to a camera which supports RTSPS would probably be best equipped to do that. Mine don't list RTSPS in the data sheet at least, and I'm unwilling to try my luck with firmware updates (if there even are any available)...

MichaIng, 2026-02-21:
Okay, I missed our connection test 😅. Here the relevant code: https://github.com/motioneye-project/motioneye/blob/main/motioneye/handlers/config.py#L423-L441

It would hence require an additional `utils/rtsps.py`, or expand `utils/rtsp.py` to support RTSPS: https://github.com/motioneye-project/motioneye/blob/main/motioneye/utils/rtsp.py

Would be actually an idea to allow blindly applying the first camera of the given URL, if we do not support the scheme directly. Hence no connection test, no explicit stream selection, just applying the URL as is to the motion camera config. But better would be a proper connection test and camera stream list to choose from, indeed.